Xilinx/finn #3: Network deployment on PYNQ

reshkor, 2018-08-21:
Hi,

I have modified existing CIFAR-10 network to detect 5 new classes and generated .bit file.
How can I deploy the network on PYNQ?
I have executed make_pynq_standalone_exe.sh on PYNQ board however it returned error:
(.text+0x28): undefined reference to `main'
collect2: error: ld returned 1 exit status
root@pynq:~/finn-build-3K8vFW# cd ..

HPC-Ken, 2018-09-03:
Hi,

You can checkout https://github.com/Xilinx/BNN-PYNQ. If you replace the cnv-pynq-pynq.bit in bnn/bitstreams/ with your bit file, the notebook will use your implementation.

reshkor, 2018-09-03:
Hello,
Thank for your response.
I have tried replacing .bit and .tcl files from FINN on BNN checkout, however this was not enough. As I understand the bnn overlay for PYNQ tries to load weight parameters from params folder, which are saved in .bin format. The parameters from FINN project are saved in several params_*.h files. 
Did you use a script to parse weight parameters to .bin file or did you change the bnn overlay driver to load parameters from .h files instead?

HPC-Ken, 2018-09-05:
Hi,

Bit files produced by this release of FINN bake the weights into the bit file itself, therefore loading weights at runtime is not necessary. If you comment out line 85 of https://github.com/Xilinx/BNN-PYNQ/blob/master/bnn/bnn.py, it will use the built in weights from the bit file.

reshkor, 2018-09-05:
Commenting line 85 and replacing .bit file did work for original CIFAR-10 and "CIFAR-5"(generated by FINN) for hw inference. However sw inference always returns first class (maybe sw does not have the weights since they are baked in .bit?). I removed number of output classes and re-trained the network for "CIFAR-5". 
I also have set up another model based on CIFAR-10 with additional 3 conv layers and reduced pooling layers to train. 
Do I need to change something extra to account for new layers (.tcl file, .so files) on PYNQ device or changing .bit file would be enough?

HPC-Ken, 2018-09-06:
PYNQ will look for a corresponding tcl file for any given bit file. If your new bit file is called cifar5.bit for example, you will need a cifar5.tcl. Copying one of the existing tcl files should work.

reshkor, 2018-09-12:
For the new CIFAR-10 with 3 conv layers model I have replaced .bit file in BNN-PYNQ and commented out loading of the parameters. The inference results for all classes are always below -0.8. Attached a file from Cifar10 example notebook.
[image]

closed 2019-02-07 by HPC-Ken