Xilinx/finn #4: Nice work and would like to explore more

eerobert, 2018-09-06:
Hi,

I have some question on this repo

1. Which vivado version is tested/supported?
2. Training should be done using HWGQ or ordinary Caffe?

Thanks

eerobert, 2018-09-07:
i think i have the answer for 1. 4. and 5.
it should be 2017.4

HPC-Ken, 2018-09-12:
1. We tested with 2017.4.
2. HWGQ only. 
3. See https://github.com/Xilinx/FINN/blob/master/FINN/test/test_hwgq_cnv_w1a1.py. It demonstrates testing a CNV network on MNIST.
4. Not tested with 2018.2. How does it fail?
5. The resource utilization of designs will vary between Vivado releases. You can adjust the amount of BRAM that the tool will attempt to use in https://github.com/Xilinx/FINN/blob/master/FINN/core/config.py. Reduce BRAM_PROPORTION to 0.9 or below.

closed 2018-09-27 by HPC-Ken