Xilinx/finn #11: Any limitation on layer array size?

eerobert, 2018-10-04:
I am trying to input a 227x227 image to the network.
Is it possible?

eerobert, 2018-10-11:
Thanks for your comment.

Apart from size limitation, do you know can I have non-square input?

reshkor, 2018-10-11:
I tried only with 32x32 images for cnv overlay. I guess you can give it any image size below 32x32 in the input layer for training. However the software managing communication to PYNQ hardware (.so libraries generated from https://github.com/Xilinx/FINN/tree/master/FINN/backend/fpga/host/bnn) is fixed and the network still will treat input on hardware as a 32x32x3 image (for cnv overlay).

eerobert, 2018-10-12:
you mean "testPrebuiltCIFAR10"? 
it seems like it is built only for CNV example

reshkor, 2018-10-12:
Oh, sorry I got confused. Do you want to create a network with input layer configured for 227x227 images or just give a 227x227 image to an existing network?

If it is the first one then (as far as I understand) it is impossible without additional changes to the network libraries. CNV is limited to 32x32 input layer and LFC is 28x28. 
I tried to create my network with input layers configured to read 64x64 images and it failed due to parameter limitation. 

If you want to pass an image to implemented network then it has to be scaled to 32x32 for CNV and 28x28 for LFC in order to work. At least that's what BNN-PYNQ does to all input images.
There is an example (CNV-BNN_Cifar10 of BNN-PYNQ) where a large image (680x480 or so) is tested. The image is sliced in several small 32x32 images and then classification is performed on the slices.

HPC-Ken, 2018-10-23:
> Hey,
> I have tried with 64x64 but it failed as the number of parameters after the first convolution layer was too big for tool to handle. The parameters number is limited to 16-bit type(thus 32k parameters). Not sure what is the reason for the limit.

What error did the tool give? There shouldn't be a limitation, unless the resource model predicts a larger footprint than will fit in the target chip.

HPC-Ken, 2018-10-23:
Non Square inputs should be supported, as the height and width parameters are distinctly handled within the tool.

reshkor, 2018-10-23:
> What error did the tool give? There shouldn't be a limitation, unless the resource model predicts a larger footprint than will fit in the target chip.

I have changed cnv-w1a1.protox input layer from 32x32 to 64x64
The error:
```
Most computationally expensive layer is #1, a FPGABipolarConvThresholdLayer with 265420800 operations.
Achieved fps of 256.821170 with 3.520729% LUT utilisation and 98.773522% BRAM utilisation
LUTS: 1873/53200, BRAM 275/279 
Traceback (most recent call last):
  File "FINN/bin/finn", line 192, in <module>
    process_args(args)
  File "FINN/bin/finn", line 94, in process_args
    generate_hardware(net, dev, gen_bitfile=False)
  File "FINN/bin/finn", line 159, in generate_hardware
    ret = fpga_backend.synthesize(hlslayers, net, dev, res_alloc_predetermined, dirpath, "sfcall-")
  File "/home/steven/FINN/FINN/backend/fpga/backend_fpga.py", line 368, in synthesize
    map(lambda x: x.codegen_params(output_dir), pipeline)
  File "/home/steven/FINN/FINN/backend/fpga/backend_fpga.py", line 368, in <lambda>
    map(lambda x: x.codegen_params(output_dir), pipeline)
  File "/home/steven/FINN/FINN/backend/fpga/layers_fpga.py", line 485, in codegen_params
    packed_t = cg.packThresholdMatrix(self.getT(), self.getPE(), self.getAccWidth())
  File "/home/steven/FINN/FINN/backend/fpga/utils.py", line 135, in packThresholdMatrix
    raise Exception("Non-int16 thresholding not yet supported.")
Exception: Non-int16 thresholding not yet supported.
```

maltanar, 2020-02-25:
I'm closing all issues relating to the v.01 version in preparation for the new release. Please note that v0.1 is now deprecated and unsupported.

closed 2020-02-25 by maltanar