Xilinx/finn #12: Example network cnv-w1a1 fails on BNN-PYNQ

reshkor, 2018-10-10:
Hello,

I have tried to implement example convolution network (cnv-w1a1.caffemodel) on PYNQ, but it always fails.
I have used BNN-PYNQ project (commits: 1365efa233fa8a66b1c9e68e2f1d00e3bf834f8b and 2fe7f6ad3968bcb4b496203b3e7e73e325d3b4a2) to deploy .bit file created by FINN.
Steps to reproduce:

1. Start PYNQ board
2. Comment out line `self.interface.load_parameters(params.encode())` in bnn/bnn.py of BNN-PYNQ.
3. Replace bnn/bitstreams/cnv-pynq-pynq.bit (or CNV-W1A1-pynq-Z1-Z2 in latest releases) with FINN generated .bit file
4. Start Cifar10 notebook example
5. Launch BNN in hardware
6. Change pictures in Launch BNN in hardware

The results for each picture will be always Deer.

HPC-Ken, 2018-10-23:
I will investigate

saraballeri, 2018-10-24:
Hi @reshkor ,
I have mostly performed your actions.
1. I have created .bit file through FINN framework
2. In bnn.py, I modified the bit file to load in the init of PynqBNN class.
3. I try to comment out `self.interface.load_parameters(params.encode())` as you say (but i don't understand why)
4. I start the cifar10 notebook example, but when i launch BNN in hardware the process blocks on `self.interface.inference(....)`
Can you explain me what are the passes that you follow?

Thanks,
Sara

reshkor, 2018-10-25:
Hello @saraballeri ,

1. I hope you didn't change layer composition in .prototxt file when creating .bit file in FINN. FINN only supports two networks.
2. What did you modify here? The only thing I did is to comment `self.interface.load_parameters(params.encode())` line https://github.com/Xilinx/BNN-PYNQ/blob/65f212439715bfd214104b23d418b167d5f840c9/bnn/bnn.py#L112
3. The line will try to load the network parameters stored in BNN-PYNQ/bnn/params folder. FINN .bit file has the parameters baked already thus loading is not needed.
4. Seems like it can't access the network parameters to make a prediction, since the commented line prevents loading BNN-PYNQ parameter files. Probably BNN-PYNQ .bit file is loaded instead of FINN .bit file (happened to me).

Slightly elaborated steps:

1. I guess you have a .bit file that complies with networks described here https://docs.google.com/presentation/d/17oorGvtUbdFd-o1OzSuxGCSrWsvm_S2ftC1UC2FLtuE/edit#slide=id.p3 slides 12 or 20. If not you can use cnv-w1a1.prototxt and cnv-w1-a1.caffemodel that are already supplied as an example and run the command to synthesize CNV network.
2. Place the generated .bit file in BNN-PYNQ/bnn/bitstreams/pynqZ1-Z2 folder and rename it to cnvW1A1-pynqZ1-Z2.bit. So instead of original BNN-PYNQ .bit file the FINN .bit file is loaded in bnn overlay. I installed BNN-PYNQ as a development module thus it was in /home/xilinx/src/bnn-dev/bnn but if you follow the instructions on BNN-PYNQ it is installed somewhere in /opt/python/../../../bnn
3. Comment the line `self.interface.load_parameters(params.encode())` in BNN-PYNQ/bnn/bnn.py
4. Run CIFAR10 example. In line 
`hw_classifier = bnn.CnvClassifier(bnn.NETWORK_CNVW1A1,'cifar10',bnn.RUNTIME_HW)` bnn.NETWORK_CNVW1A1 will load cnvW1A1-pynqZ1-Z2.bit from bitstreams folder, which should be the FINN generated .bit file.

NOTE: If you first run CIFAR10 notebook example with default settings(so loading parameters from BNN-PYNQ/bnn/params folder and original cnvW1A1-pynqZ1-Z2.bit) and then make changes to load the FINN .bit file without resetting the board, the CIFAR10 notebook example will be able to classify pictures correctly. However after the reset of the Pynq board all pictures will be classified as 'Deer' with FINN .bit file.

Hope it helps

saraballeri, 2018-10-25:
Hi @reshkor ,
thank you for your quick response and your full explanation.
I'll try again to perform those steps.
Anyway:
1. Is it possible to use binarized networks like AlexNet, GoogleNet, etc. provided by [HWGQ Caffe](https://github.com/zhaoweicai/hwgq)? 
2. Or can I only use cnvw1a1 and lfcw1a1 networks on FINN?
3. How can I provide to FINN an other binarized network different from cnvw1a1 and lfcw1a1? 

Sorry for the lot of questions, but I'm doing my master thesis and I'm coming close to the world of neural networks on FPGA now.

Thanks a lot,
Sara

reshkor, 2018-10-25:
@saraballeri ,

1. Those networks are not supported by FINN and BNN-PYNQ, only cnv and lfc are.
2. You can re-use structures of cnv and lfc (basicly .prototxt files) to train networks to detect different objects. In the .prototxt you can change number of output classes (last fully connected layer). For BNN-PYNQ it is limited to 64 not sure if it is the same for FINN. 
You can create your binarized network with custom number of layers but you would also have to add support for the new overlay in BNN-PYNQ (check out power-point presentation I linked in previous post, it gives some tips from slide 27). However it seems like quite a bit of work and as I'm also writing master thesis I don't have much time to spend on implementing support for new network overlays.

saraballeri, 2018-10-25:
@reshkor ,
exactly the same things for me.
I cannot spend much time on implementing new overlays :(
Thanks very much, your explanation has removed all doubt from my mind.

Sara

maltanar, 2020-02-25:
I'm closing all issues relating to the v.01 version in preparation for the new release. Please note that v0.1 is now deprecated and unsupported.

closed 2020-02-25 by maltanar