Xilinx/finn #13: Question about layers of cnv network

reshkor, 2018-10-11:
Hello,

I have two questions regarding the layers of cnv network.

1.  Why is there a max-pool layer after the last 256 filter convolution layer (conv6)?
As I understand there is no padding in the convolution layers thus the inputs for layers would be: 32x32(conv1)->30x30(for conv2)->28x28(maxpool1)->14x14(conv3)->12x12(conv4)->10x10(max-pool2)->5x5(conv5)->3x3(conv6)->1x1(max-pool3)->1x1(FullyConnected1). Isn't max-pool3 redundant since there is only one input?

2.  Why there is no Softmax layer placed at the end? Wouldn't it make the end results clearly defined?

Thank you!

reshkor, 2018-10-12:
I refer to FINN CNV https://github.com/Xilinx/FINN/blob/master/FINN/inputs/cnv-w1a1.prototxt lines 389-398.

maltanar, 2020-02-25:
I'm closing all issues relating to the v.01 version in preparation for the new release. Please note that v0.1 is now deprecated and unsupported.

closed 2020-02-25 by maltanar