Xilinx/finn #16: HWGQ Caffe is necessary for deployment?

saraballeri, 2018-10-23:
Hi,
I would use your framework only for running inference on networks. Is it necessary to install HWGQ Caffe? I'm just going to use prototxt and caffemodel provided.
Thanks,
Sara

HPC-Ken, 2018-10-23:
Hi Sara,

Yes, it's necessary to install HWGQ as we use it's python bindings to handle prototxt parsing and layer attribute extraction. 

Ken

saraballeri, 2018-10-23:
Ok, thanks for your quick response.
Another question, is it then possible to run inference on any networks listed above section 'Models' of this repo: https://github.com/zhaoweicai/hwgq? (in particular for example I think to have only the necessity to download AlexNet_HWGQ_BW. caffemodel and its deploy.prototxt) 
Thanks,

Sara

HPC-Ken, 2019-02-07:
FINN is only tested with the LFC and CNV models which we provide. Compatibility with all HWGQ networks is not guaranteed.

closed 2019-02-07 by HPC-Ken