Xilinx/finn #17: How to run inference from python interface

saraballeri, 2018-10-24:
Hi,
I have succesfully generated the bitstream and the tcl file on my own executing the following command (suggested in the guide of this repo):
"python FINN/bin/finn --device=pynqz1 --prototxt=FINN/inputs/cnv-w1a1.prototxt --caffemodel=FINN/inputs/cnv-w1a1.caffemodel --mode=synth"

I thought to load the bitstream on PYNQ-Z1 and try to run inference.
Can I use BNN-PYNQ (https://github.com/Xilinx/BNN-PYNQ), changing the loading of bitstream, to make inference of any network or you provide a python interface, like BNN-PYNQ, to run inference on a network?
 
Thanks,
Sara

nmoroze, 2019-04-19:
Any updates on this?

HPC-Ken, 2019-04-19:
Yes, you can use the BNN-PYNQ repo for deploying the bit file. 

In https://github.com/Xilinx/BNN-PYNQ/blob/master/bnn/bnn.py you'll need to comment out "self.bnn.load_parameters(os.path.join(params, network))"  on line 183 or 293 depending on the network you're deploying. 

The reason being, the FINN tool bakes the weights into the bit file, and the BNN-PYNQ assumes they are loaded at runtime.

maltanar, 2020-02-25:
I'm closing all issues relating to the v.01 version in preparation for the new release. Please note that v0.1 is now deprecated and unsupported.

closed 2020-02-25 by maltanar