Xilinx/finn #18: Implementation fail using the CNV network

FrancoCapraro, 2018-10-30:
Hi, i do the estimate and synthesize CNV network **as the README.md say in Line 28**:
`
python FINN/bin/finn --device=pynqz1 --prototxt=FINN/inputs/cnv-w1a1.prototxt --caffemodel=FINN/inputs/cnv-w1a1.caffemodel --mode=synth
`

But i coudnt see the implementation. Im new with this and i cant find my error.
(I have Caffe , i did the test,runtest,etc) and the others Prerequisites.

**Here is the terminal text:**
`
Time (s): cpu = 00:00:39 ; elapsed = 00:00:22 . Memory (MB): peak = 3556.273 ; gain = 0.000 ; free physical = 5047 ; free virtual = 20775
INFO: [Common 17-83] Releasing license: Implementation
60 Infos, 21 Warnings, 0 Critical Warnings and 5 Errors encountered.
place_design failed
ERROR: [Common 17-69] Command failed: Placer could not place all instances
INFO: [Common 17-206] Exiting Vivado at Tue Oct 30 12:48:13 2018...
[Tue Oct 30 12:48:14 2018] impl_1 finished
wait_on_run: Time (s): cpu = 00:21:10 ; elapsed = 00:22:14 . Memory (MB): peak = 1659.152 ; gain = 0.000 ; free physical = 5081 ; free virtual = 20809
INFO: [Common 17-206] Exiting Vivado at Tue Oct 30 12:48:14 2018...
cp: cannot stat '/tmp/finn-build-v4KuS_/finnaccel/finnaccel.runs/impl_1/procsys_wrapper.bit': No such file or directory
Traceback (most recent call last):
  File "FINN/bin/finn", line 192, in <module>
    process_args(args)
  File "FINN/bin/finn", line 92, in process_args
    generate_hardware(net, dev, gen_bitfile=True)
  File "FINN/bin/finn", line 161, in generate_hardware
    ret.synthesis_bitfile()	
  File "/home/franco/Downloads/GIT/FINN/FINN/backend/fpga/backend_util.py", line 71, in synthesis_bitfile
    subprocess.check_call(["sh", self.getBitfileSynthScriptPath()], cwd=self.path)
  File "/usr/lib/python2.7/subprocess.py", line 541, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['sh', '/tmp/finn-build-v4KuS_/make_pynq_bitfile.sh']' returned non-zero exit status 1
`

Fully Implementation text:
[https://textuploader.com/dw3uo](url)

FrancoCapraro, 2018-11-08:
I find my error.

closed 2018-11-08 by FrancoCapraro

pmousoul, 2019-03-29:
@FrancoCapraro 
What is your solution - I'm facing the same problem!