Xilinx/finn #22: Non-int16 thresholding not yet supported

eejlny, 2018-11-26:
Hello,

I have been trying to do some simple experiments trying to create other networks using the hwgq tool and then asking FINN to create hardware for it but I am not getting very far with errors like with a binarized alexnet from hwgq: Exception("Threshold does not preserve max")

To try something simple I used the provided example for CIFAR CNV W1A1 and increase the width by doubling the number of channels in each layer. This fits perfectly in a VU9P device but when doing :
 python FINN/bin/finn --device=vu9p --prototxt=FINN/inputs/alexnet2.prototxt  --mode=estimate
I get : 
Exception: Non-int16 thresholding not yet supported.
can someone clarify if something can be done about these errors ?

eejlny, 2018-11-27:
Here is the prototxt that fails with the Non-int16 thresholding not yet supported.. 

[alexnet2.txt](https://github.com/Xilinx/FINN/files/2618976/alexnet2.txt)

maltanar, 2020-02-25:
I'm closing all issues relating to the v.01 version in preparation for the new release. Please note that v0.1 is now deprecated and unsupported.

closed 2020-02-25 by maltanar