Xilinx/finn #24: Building the SDx

FrancoCapraro, 2018-12-05:
Hi. im following the steps in FINN/FINN/backend/fpga/README.md .
Im doing the build of SDx executing this steps:

1) " ./make-sw.sh lfc-max bnn hlsweights" choosing HLS (h) [OK]
2) "./example/lfc-max_bnn_hlsweights/rawhls-lfc-max" choosing 10 and 1 as test. [OK]
3)" ./make-sw.sh lfc-max bnn sdx "  choosing SDx (x) [ERROR]

`HLS Simulation or host for HW accelerator or SDx (S/H/X)? x
Building host
0 Building xclbin
make SDA_FLOW=hw run_hw_int -f sdaccel.mk
make[1]: Entering directory '/home/franco/Downloads/GIT/FINN/FINN/backend/fpga/scripts'
make SDA_FLOW=hw xbin -f sdaccel.mk
make[2]: Entering directory '/home/franco/Downloads/GIT/FINN/FINN/backend/fpga/scripts'
make[2]: Nothing to be done for 'xbin'.
make[2]: Leaving directory '/home/franco/Downloads/GIT/FINN/FINN/backend/fpga/scripts'
source setup.sh;/ ../BlackBoxJam_hw.xclbin
**/bin/bash: setup.sh: No such file or directory**
/bin/bash: /: Is a directory
common.mk:106: recipe for target 'run_hw_int' failed
make[1]: *** [run_hw_int] Error 126
make[1]: Leaving directory '/home/franco/Downloads/GIT/FINN/FINN/backend/fpga/scripts'
common.mk:100: recipe for target 'run_hw' failed
make: *** [run_hw] Error 2
Output file: /home/franco/Downloads/GIT/FINN/FINN/backend/fpga/example/lfc-max_bnn_sdx//sdx-lfc-max
Done!
`

So tried commenting the line 104 in _/FINN/FINN/backend/fpga/scripts/common.mk_ because i dont see any setup.sh ( i have the source of SDx  **source /opt/Xilinx/SDxSoc/SDx/2018.2/settings64.sh** ) in the folders:

`run_hw_int : host xbin_hw
#	source ${BOARD_SETUP_FILE};${HOST_EXE_DIR}/${HOST_EXE} ${HOST_ARGS}`

without that line i got:

`HLS Simulation or host for HW accelerator or SDx (S/H/X)? x
Building host
0 Building xclbin
make SDA_FLOW=hw run_hw_int -f sdaccel.mk
make[1]: Entering directory '/home/franco/Downloads/GIT/FINN/FINN/backend/fpga/scripts'
make SDA_FLOW=hw xbin -f sdaccel.mk
make[2]: Entering directory '/home/franco/Downloads/GIT/FINN/FINN/backend/fpga/scripts'
make[2]: Nothing to be done for 'xbin'.
make[2]: Leaving directory '/home/franco/Downloads/GIT/FINN/FINN/backend/fpga/scripts'
make[1]: Leaving directory '/home/franco/Downloads/GIT/FINN/FINN/backend/fpga/scripts'
Output file: /home/franco/Downloads/GIT/FINN/FINN/backend/fpga/example/lfc-max_bnn_sdx//sdx-lfc-max
Done!
`

But if i do the execution:

`fpga git:(f4e48af) ✗ ./example/lfc-max_bnn_sdx/sdx-lfc-max
Building network...
Setting network weights and thresholds in accelerator...
Enter number of images from MNIST test set to check:
10
Running prebinarized test for 10 images...
Enter number of times to repeat test: 
1
ERROR: No devices found
Failed to detect platform
[1]    10191 segmentation fault (core dumped)  ./example/lfc-max_bnn_sdx/sdx-lfc-max
`

HPC-Ken, 2018-12-05:
Hi, 

What platform are you attempting to run this on?

FrancoCapraro, 2018-12-05:
hi , thanks for your reply. Im using Ubuntu 16.04

HPC-Ken, 2018-12-05:
Hi, sorry, I meant what physical hardware are you trying to run on. What FPGA?

FrancoCapraro, 2018-12-05:
ohh , my bad. Well, i choose ku115 (becuase the opcion is "where <platform> = ku115 zc706" ) but i want to do a porting to Ultra96.

HPC-Ken, 2018-12-05:
Ok, you've been able to compile but when you try to execute it can't find a ku115 on your system.

I should point out that this release is only applicable to PYNQ platforms. The SDx flow is not supported. Also, I don't believe SDx applies to Ultra96. Any of the SDx code here is for datacenter execution only.

You should use the Python frontend in the top level of this repository. The tool currently doesn't support Ultra96 but you could make a new file in /devices which specifies the Ultra96 and maybe it will implement a bitstream for you.

FrancoCapraro, 2018-12-05:
ok  thanks for your time !

FrancoCapraro, 2018-12-06:
So. i do the porting to the Ultra96 and get the bitstream, but now my problem is the Aplication to use it

closed 2019-02-20 by HPC-Ken