Xilinx/finn #25: Error while running FINN in  estimate mode

jayanth517, 2019-01-27:
Hi,
I am trying to understand FINN framework and I have tried to run the example mentioned in the FINN's github page. The example I have tried is to estimate performance of LFC MLP network and I have run into the following error. I kindly request you to please help me understand the reason for the error and also can you please let me know which directory should i see for output of the flow in the estimate. My use case is to get an estimate of no of LUT's or hardware modules for the example BNN network.

FYI :- I have cloned the HWGQ repo(https://github.com/zhaoweicai/hwgq) and installed caffe following the instruction from the caffe page(http://caffe.berkeleyvision.org/installation.html) 

**Error** :-
jayanth5178@bohr3:~/mywork/finn/FINN$ python FINN/bin/finn --device=pynqz1 --prototxt=FINN/inputs/lfc-w1a1.prototxt --mode=estimate
[libprotobuf ERROR google/protobuf/text_format.cc:298] Error parsing text-format caffe.NetParameter: 32:15: Message type "caffe.LayerParameter" has no field named "quant_param".
WARNING: Logging before InitGoogleLogging() is written to STDERR
F0122 06:25:26.846015  3394 upgrade_proto.cpp:90] Check failed: ReadProtoFromTextFile(param_file, param) Failed to parse NetParameter file: FINN/inputs/lfc-w1a1.prototxt
*** Check failure stack trace: ***
Aborted (core dumped)

HPC-Ken, 2019-02-11:
Docker image resolves dependency issue

HPC-Ken, 2019-02-26:
To run in synth mode you need to have vivado installed with a valid license.

HPC-Ken, 2019-02-26:
Your custom LSTM can only be used in FINN if it is expressed in HWGQ Caffe

closed 2019-03-14 by HPC-Ken