Xilinx/finn #26: Problem with Vivado

Sayf-Khan, 2019-02-17:
Hello,

I am trying to test the following command:
"python FINN/bin/finn --device=pynqz1 --prototxt=FINN/inputs/cnv-w1a1.prototxt --caffemodel=FINN/inputs/cnv-w1a1.caffemodel --mode=synth"

However I am having problems getting it to run, here are the details from the terminal:
"
_Most computationally expensive layer is #1, a FPGABipolarConvThresholdLayer with 57802752 operations.
Achieved fps of 7086.167800 with 11.216347% LUT utilisation and 54.239471% BRAM utilisation
LUTS: 5967/53200, BRAM 151/279 
Outputting to:  /tmp/finn-build-xbtloK
/tmp/finn-build-xbtloK/make_pynq_bitfile.sh: 11: /tmp/finn-build-xbtloK/make_pynq_bitfile.sh: vivado_hls: not found
/tmp/finn-build-xbtloK/make_pynq_bitfile.sh: 13: /tmp/finn-build-xbtloK/make_pynq_bitfile.sh: vivado: not found
cp: cannot stat '/tmp/finn-build-xbtloK/finnaccel/finnaccel.runs/impl_1/procsys_wrapper.bit': No such file or directory
Traceback (most recent call last):
  File "FINN/bin/finn", line 192, in <module>
    process_args(args)
  File "FINN/bin/finn", line 92, in process_args
    generate_hardware(net, dev, gen_bitfile=True)
  File "FINN/bin/finn", line 161, in generate_hardware
    ret.synthesis_bitfile()	
  File "/app/FINN/FINN/backend/fpga/backend_util.py", line 71, in synthesis_bitfile
    subprocess.check_call(["sh", self.getBitfileSynthScriptPath()], cwd=self.path)
  File "/usr/lib/python2.7/subprocess.py", line 190, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['sh', '/tmp/finn-build-xbtloK/make_pynq_bitfile.sh']' returned non-zero exit status 1
root@4d94b8fd7427:/app/FINN# 
"_

From my understanding, this is to do with Vivado not being located.
I'm using Linux Ubuntu 18.04 with Vivado 2018.2 installed.

I am very new to linux, so I believe it is my lack of knowledge in this area where I am really struggling on.

Kind regards

HPC-Ken, 2019-02-19:
Hi, 

If you have vivado installed, ensure you have it in your path. Run "which vivado" and it should show you that vivado is in your path. If not, you should source the setup.sh file in your install.

Best Regards,

Ken

Sayf-Khan, 2019-02-24:
Good afternoon, Thank you for your help from before and I was able to do what you said however vivado still can't be launched from FINN even though I have added the settings64.sh to PATH.
As you can see here
sayf@sayf-MS-7B61:~$ $PATH bash: /home/sayf/bin:/home/sayf/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/opt/Xilinx/Vivado/2018.3/settings64.sh

To test this I type in which vivado however there is no output from the terminal as seen here: sayf@sayf-MS-7B61:~$ which vivado 
sayf@sayf-MS-7B61:~$

However when I source it directly, it gives me this:
sayf@sayf-MS-7B61:/$ source /opt/Xilinx/Vivado/2018.3/settings64.sh 
sayf@sayf-MS-7B61:/$ which vivado /opt/Xilinx/Vivado/2018.3/bin/vivado

Which shows it is working.

Kind regards, Sayf Khan

HPC-Ken, 2019-02-26:
Hi, 

Don't add the script to your PATH. Source it directly as you have done. Does this resolve your issue?

Ken

closed 2019-03-14 by HPC-Ken