Xilinx/finn #29: Failed to parse NetParameter file

KarimAli2017, 2019-02-27:
Hello 
When I tried to synthesize one of the network examples
root@d9322fb83d76:/app/FINN# python FINN/bin/finn --device=pynqz1 --prototxt=FINN/inputs/sfc-w1a1.prototxt --caffemodel=FINN/inputs/sfc-w1a1.caffemodel --mode=synth

I got that error

WARNING: Logging before InitGoogleLogging() is written to STDERR
F0227 14:36:23.989655    52 upgrade_proto.cpp:95] Check failed: ReadProtoFromBinaryFile(param_file, param) Failed to parse NetParameter file: FINN/inputs/sfc-w1a1.caffemodel
*** Check failure stack trace: ***
Aborted (core dumped)

Any recommendations ? 
noting vivado tools are installed

HPC-Ken, 2019-02-27:
Did you install git lfs and run git lfs pull?

KarimAli2017, 2019-02-27:
I run it but I got that 
Outputting to:  /tmp/finn-build-nvE5gH
/tmp/finn-build-nvE5gH/make_pynq_bitfile.sh: 11: /tmp/finn-build-nvE5gH/make_pynq_bitfile.sh: vivado_hls: not found
/tmp/finn-build-nvE5gH/make_pynq_bitfile.sh: 13: /tmp/finn-build-nvE5gH/make_pynq_bitfile.sh: vivado: not found
cp: cannot stat '/tmp/finn-build-nvE5gH/finnaccel/finnaccel.runs/impl_1/procsys_wrapper.bit': No such file or directory
Traceback (most recent call last):
  File "FINN/bin/finn", line 192, in <module>
    process_args(args)
  File "FINN/bin/finn", line 92, in process_args
    generate_hardware(net, dev, gen_bitfile=True)
  File "FINN/bin/finn", line 161, in generate_hardware
    ret.synthesis_bitfile()	
  File "/app/FINN/FINN/backend/fpga/backend_util.py", line 71, in synthesis_bitfile
    subprocess.check_call(["sh", self.getBitfileSynthScriptPath()], cwd=self.path)
  File "/usr/lib/python2.7/subprocess.py", line 190, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['sh', '/tmp/finn-build-nvE5gH/make_pynq_bitfile.sh']' returned non-zero exit status 1

Should I execute that command docker run -t finn everytime I open the framework ?
How could I add Vivado at that moment ?

HPC-Ken, 2019-02-27:
Vivado and Vivado HLS are not in your path. You need to source the settings files for each of them.

KarimAli2017, 2019-02-28:
Could you please tell me how can I source them while opening the docker container ?

HPC-Ken, 2019-03-14:
You can install the tools in your container, or mount the executable into the container from the host:

https://stackoverflow.com/questions/30362373/run-executable-from-host-within-docker-container

closed 2019-03-14 by HPC-Ken