Xilinx/finn #31: XNOR-popcount

jdwo, 2019-03-05:
Hello,

I have a basic question to the implementation of the dot product. I try to comprehend the XNOR-popcount operation on an example.

1 -1  1 -1  1  1
1  1 -1  1  1 -1

The result should be -2. By simple using XNOR and popcount I get a result of 2.

I found the formel
2*p-N
which gives exactly the expected -2 with the XNOR-popcount result. Did I overlook something in the FINN-Paper? I could not find an explanation in the paper. Maybe I misunderstood something completely.

Thank you for your help

Kind regards

maltanar, 2019-03-05:
This is due to the fact that we represent -1 values using a zero bit (i.e. unset bits represent -1). Popcount simply counts the number of set bits (1s) in a bit vector, hence it will always return a nonnegative number. To convert the XNOR-popcount result into the dot product result, we take advantage of the fact that the bit vector can only contain set bits or unset bits: the number of set bits plus the number of unset bits will always be equal to the vector length.

In slightly more mathematical form: after we have performed XNOR, let p be the number of set bits in the bit vector (i.e. popcount result),  let n be the number of unset bits in the bit vector, and N be the vector length. We know that p+n=N, and would like to compute the dot product result p-n (since each unset bit will contribute -1, they contribute -n in total). Knowing only p and N, we can substitute n=N-p, so p-n = p - (N-p) = 2*p -N

jdwo, 2019-03-05:
Thank you very much for the explanation.

closed 2019-03-05 by jdwo

HyunwooKim2, 2019-05-22:
Hello,

I have the same questions. I understood what Mr. maltanar said.

However, the HLS source code appears to have no subtraction of unset bits from accumulation except for the first layer. From the second layer, XnorMul type is used. In the mul function (mac.hpp), the * operator of that type only performs a simple XNOR operation. In mac function (mac.hpp), only accumulation is performed with + = operation. After that, thresholding is done with activate function (mvau.hpp).

In the FINN article, I think that the threshold is also scaled by the (th + S) / 2 operation because the accumulation value is between 0 and S, where the negative is not taken into consideration. I wonder if the 2 * p-N operation mentioned by Mr. maltanar is applied to the HLS code.

Thank you.

maltanar, 2019-05-22:
You are right, we don't explicitly do the 2*p - N but instead exploit the flexibility of the thresholds to compensate for this. When doing code generation at the output of a bipolar-by-bipolar (meaning both operands are +-1) FC or conv layer, this is what gets called:

https://github.com/Xilinx/FINN/blob/master/FINN/backend/fpga/layers_fpga.py#L222

..which calls the makePositiveThresholds function here:

https://github.com/Xilinx/FINN/blob/master/FINN/backend/fpga/utils.py#L40

which is doing (th + fan-in) / 2 as you said.

HyunwooKim2, 2019-05-23:
Thank you for your quick response.
XNOR (1,0 representation) and popcount cause accumulation value to be between 0 and fanin.
And the (th+fan-in)/2 operation changes the threshold to the same scale.
Accumulation and activation can be done with popcount and makePositiveThresholds without having to do 2*p-N as you said.
I understand.
Thank you.

HyunwooKim2, 2019-05-24:
Hellow.
I have one more question.

In the source code of the link below, when the threshold is negative, you have replaced the threshold with fanin.
[https://github.com/Xilinx/FINN/blob/e6d44508b4f7c6ac8e2c6929f2ffa85ea5761708/FINN/backend/fpga/utils.py#L58](url)

I wonder why you replaced it with fan-in instead of zero.
In my opinion, if the threshold is negative, it should all be activated.

Thanks in advance for your reply.