Xilinx/finn #32: cannot stat file when running the repo synthesis example

pmousoul, 2019-03-29:
When running:

`
$ python FINN/bin/finn --device=pynqz1 --prototxt=FINN/inputs/cnv-w1a1.prototxt --caffemodel=FINN/inputs/cnv-w1a1.caffemodel --mode=synth
`

I get the following error:

`
cp: cannot stat '/tmp/finn-build-DpC6wt/finnaccel/finnaccel.runs/impl_1/procsys_wrapper.bit': No such file or directory
Traceback (most recent call last):
  File "FINN/bin/finn", line 192, in <module>
    process_args(args)
  File "FINN/bin/finn", line 92, in process_args
    generate_hardware(net, dev, gen_bitfile=True)
  File "FINN/bin/finn", line 161, in generate_hardware
    ret.synthesis_bitfile()	
  File "/mnt/terabyte/pmousoul_data/sw/FINN/FINN/backend/fpga/backend_util.py", line 71, in synthesis_bitfile
    subprocess.check_call(["sh", self.getBitfileSynthScriptPath()], cwd=self.path)
  File "/usr/lib/python2.7/subprocess.py", line 541, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['sh', '/tmp/finn-build-DpC6wt/make_pynq_bitfile.sh']' returned non-zero exit status 1
`

Any idea on how to resolve this?

HPC-Ken, 2019-04-01:
This looks like Vivado isn't in your path. Can you confirm that it is?

pmousoul, 2019-04-01:
Hello,

before running "sh env.sh" in FINN root directory 

$ echo $PATH
/usr/local/cuda-9.2/bin:/mnt/terabyte/pmousoul_data/sw/Xilinx/petalinux/tools/linux-i386/petalinux/bin:/mnt/terabyte/pmousoul_data/sw/Xilinx/petalinux/tools/common/petalinux/bin:/mnt/terabyte/pmousoul_data/sw/Xilinx/petalinux/tools/linux-i386/gcc-arm-none-eabi-r5/bin:/mnt/terabyte/pmousoul_data/sw/Xilinx/petalinux/tools/linux-i386/microblaze-xilinx-elf/bin:/mnt/terabyte/pmousoul_data/sw/Xilinx/petalinux/tools/linux-i386/microblazeel-xilinx-linux-gnu/bin:/mnt/terabyte/pmousoul_data/sw/Xilinx/petalinux/tools/linux-i386/gcc-arm-none-eabi/bin:/mnt/terabyte/pmousoul_data/sw/Xilinx/petalinux/tools/linux-i386/gcc-arm-linux-gnueabi/bin:/mnt/terabyte/pmousoul_data/sw/Xilinx/petalinux/tools/linux-i386/aarch64-none-elf/bin:/mnt/terabyte/pmousoul_data/sw/Xilinx/petalinux/tools/linux-i386/aarch64-linux-gnu/bin:/mnt/terabyte/pmousoul_data/sw/Xilinx/Vivado/2017.2/bin:/mnt/terabyte/pmousoul_data/sw/Xilinx/Vivado_HLS/2017.2/bin:/mnt/terabyte/pmousoul_data/sw/Xilinx/DocNav:/mnt/terabyte/pmousoul_data/sw/Xilinx/SDK/2017.2/bin:/mnt/terabyte/pmousoul_data/sw/Xilinx/SDK/2017.2/gnu/microblaze/lin/bin:/mnt/terabyte/pmousoul_data/sw/Xilinx/SDK/2017.2/gnu/arm/lin/bin:/mnt/terabyte/pmousoul_data/sw/Xilinx/SDK/2017.2/gnu/microblaze/linux_toolchain/lin64_be/bin:/mnt/terabyte/pmousoul_data/sw/Xilinx/SDK/2017.2/gnu/microblaze/linux_toolchain/lin64_le/bin:/mnt/terabyte/pmousoul_data/sw/Xilinx/SDK/2017.2/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin:/mnt/terabyte/pmousoul_data/sw/Xilinx/SDK/2017.2/gnu/aarch32/lin/gcc-arm-none-eabi/bin:/mnt/terabyte/pmousoul_data/sw/Xilinx/SDK/2017.2/gnu/aarch64/lin/aarch64-linux/bin:/mnt/terabyte/pmousoul_data/sw/Xilinx/SDK/2017.2/gnu/aarch64/lin/aarch64-none/bin:/mnt/terabyte/pmousoul_data/sw/Xilinx/SDK/2017.2/gnu/armr5/lin/gcc-arm-none-eabi/bin:/mnt/terabyte/pmousoul_data/sw/Xilinx/SDK/2017.2/tps/lnx64/cmake-3.3.2/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games

after running "sh env.sh" in FINN root directory

$ echo $PATH
/usr/local/cuda-9.2/bin:/mnt/terabyte/pmousoul_data/sw/Xilinx/petalinux/tools/linux-i386/petalinux/bin:/mnt/terabyte/pmousoul_data/sw/Xilinx/petalinux/tools/common/petalinux/bin:/mnt/terabyte/pmousoul_data/sw/Xilinx/petalinux/tools/linux-i386/gcc-arm-none-eabi-r5/bin:/mnt/terabyte/pmousoul_data/sw/Xilinx/petalinux/tools/linux-i386/microblaze-xilinx-elf/bin:/mnt/terabyte/pmousoul_data/sw/Xilinx/petalinux/tools/linux-i386/microblazeel-xilinx-linux-gnu/bin:/mnt/terabyte/pmousoul_data/sw/Xilinx/petalinux/tools/linux-i386/gcc-arm-none-eabi/bin:/mnt/terabyte/pmousoul_data/sw/Xilinx/petalinux/tools/linux-i386/gcc-arm-linux-gnueabi/bin:/mnt/terabyte/pmousoul_data/sw/Xilinx/petalinux/tools/linux-i386/aarch64-none-elf/bin:/mnt/terabyte/pmousoul_data/sw/Xilinx/petalinux/tools/linux-i386/aarch64-linux-gnu/bin:/mnt/terabyte/pmousoul_data/sw/Xilinx/Vivado/2017.2/bin:/mnt/terabyte/pmousoul_data/sw/Xilinx/Vivado_HLS/2017.2/bin:/mnt/terabyte/pmousoul_data/sw/Xilinx/DocNav:/mnt/terabyte/pmousoul_data/sw/Xilinx/SDK/2017.2/bin:/mnt/terabyte/pmousoul_data/sw/Xilinx/SDK/2017.2/gnu/microblaze/lin/bin:/mnt/terabyte/pmousoul_data/sw/Xilinx/SDK/2017.2/gnu/arm/lin/bin:/mnt/terabyte/pmousoul_data/sw/Xilinx/SDK/2017.2/gnu/microblaze/linux_toolchain/lin64_be/bin:/mnt/terabyte/pmousoul_data/sw/Xilinx/SDK/2017.2/gnu/microblaze/linux_toolchain/lin64_le/bin:/mnt/terabyte/pmousoul_data/sw/Xilinx/SDK/2017.2/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin:/mnt/terabyte/pmousoul_data/sw/Xilinx/SDK/2017.2/gnu/aarch32/lin/gcc-arm-none-eabi/bin:/mnt/terabyte/pmousoul_data/sw/Xilinx/SDK/2017.2/gnu/aarch64/lin/aarch64-linux/bin:/mnt/terabyte/pmousoul_data/sw/Xilinx/SDK/2017.2/gnu/aarch64/lin/aarch64-none/bin:/mnt/terabyte/pmousoul_data/sw/Xilinx/SDK/2017.2/gnu/armr5/lin/gcc-arm-none-eabi/bin:/mnt/terabyte/pmousoul_data/sw/Xilinx/SDK/2017.2/tps/lnx64/cmake-3.3.2/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games

it seems that "/mnt/terabyte/pmousoul_data/sw/Xilinx/Vivado/2017.2/bin" exists in the PATH

Thank you for your time,
Panos

HPC-Ken, 2019-04-02:
Did vivado run? It would have taken a while to produce the bit file. Or did this error show up immediately?

pmousoul, 2019-04-02:
[output.log](https://github.com/Xilinx/FINN/files/3034704/output.log)

From the attached file, it seems that Vivado runs but the bit file is not generated.

HPC-Ken, 2019-04-02:
Looks like it fails. You could try to find the specific error in the vivado logs as to why it doesn't produce a bit file.

pmousoul, 2019-04-04:
[vivado.log](https://github.com/Xilinx/FINN/files/3045631/vivado.log)

There is no error in the vivado.log found in "/tmp/finn-build-JoqpZl/" which is attached.

Also, none of the log files referenced at the end of the vivado.log file exist:

`[Thu Apr  4 21:55:42 2019] Launched procsys_auto_pc_0_synth_1, procsys_auto_pc_1_synth_1, procsys_rst_ps7_100M_0_synth_1, procsys_BlackBoxJam_0_0_synth_1, procsys_ps7_0_synth_1, synth_1...
Run output will be captured here:
procsys_auto_pc_0_synth_1: /tmp/finn-build-JoqpZl/finnaccel/finnaccel.runs/procsys_auto_pc_0_synth_1/runme.log
procsys_auto_pc_1_synth_1: /tmp/finn-build-JoqpZl/finnaccel/finnaccel.runs/procsys_auto_pc_1_synth_1/runme.log
procsys_rst_ps7_100M_0_synth_1: /tmp/finn-build-JoqpZl/finnaccel/finnaccel.runs/procsys_rst_ps7_100M_0_synth_1/runme.log
procsys_BlackBoxJam_0_0_synth_1: /tmp/finn-build-JoqpZl/finnaccel/finnaccel.runs/procsys_BlackBoxJam_0_0_synth_1/runme.log
procsys_ps7_0_synth_1: /tmp/finn-build-JoqpZl/finnaccel/finnaccel.runs/procsys_ps7_0_synth_1/runme.log
synth_1: /tmp/finn-build-JoqpZl/finnaccel/finnaccel.runs/synth_1/runme.log
[Thu Apr  4 21:55:42 2019] Launched impl_1...
Run output will be captured here: /tmp/finn-build-JoqpZl/finnaccel/finnaccel.runs/impl_1/runme.log
launch_runs: Time (s): cpu = 00:00:09 ; elapsed = 00:00:09 . Memory (MB): peak = 1444.164 ; gain = 119.699 ; free physical = 7816 ; free virtual = 16449
# wait_on_run impl_1
[Thu Apr  4 21:55:42 2019] Waiting for impl_1 to finish...
[Thu Apr  4 21:56:00 2019] impl_1 finished
wait_on_run: Time (s): cpu = 00:00:16 ; elapsed = 00:00:18 . Memory (MB): peak = 1444.164 ; gain = 0.000 ; free physical = 6181 ; free virtual = 14815
INFO: [Common 17-206] Exiting Vivado at Thu Apr  4 21:56:00 2019...`

HPC-Ken, 2019-04-08:
So the reason that they would not exist is that there is a failure before they are written. You could zip up the files and link me to them.

pmousoul, 2019-04-12:
There you go:
https://drive.google.com/open?id=1-yHFciAnaovpKqarPjCs0CGiA6QpHjBw

HPC-Ken, 2019-04-19:
I've run the tool with your command using Vivado 2018.3 and it works fine for me. For you it's failing to generate "procsys_wrapper.bit". Try deleting the hls_syn and finnaccel directories and running  ./make_pynq_bitfile again.

pmousoul, 2019-04-21:
Hello,
I did what you said and I got:
``
# wait_on_run impl_1
[Sun Apr 21 21:19:27 2019] Waiting for impl_1 to finish...
[Sun Apr 21 21:19:42 2019] impl_1 finished
wait_on_run: Time (s): cpu = 00:00:14 ; elapsed = 00:00:15 . Memory (MB): peak = 1445.414 ; gain = 0.000 ; free physical = 4931 ; free virtual = 16481
INFO: [Common 17-206] Exiting Vivado at Sun Apr 21 21:19:42 2019...
cp: cannot stat '/tmp/finn-build-E1vW3P/finnaccel/finnaccel.runs/impl_1/procsys_wrapper.bit': No such file or directory
``

using vivado 2017.2.

I got a more elaborate output related to the problem with vivado 2018.2:
``
Starting Placer Task
INFO: [Place 46-5] The placer was invoked with the 'ExtraTimingOpt' directive.
INFO: [Place 30-611] Multithreading enabled for place_design using a maximum of 8 CPUs

Phase 1 Placer Initialization

Phase 1.1 Placer Initialization Netlist Sorting
Netlist sorting complete. Time (s): cpu = 00:00:00.06 ; elapsed = 00:00:00.07 . Memory (MB): peak = 3568.453 ; gain = 0.000 ; free physical = 5577 ; free virtual = 15788
Phase 1.1 Placer Initialization Netlist Sorting | Checksum: f8841624

Time (s): cpu = 00:00:00.07 ; elapsed = 00:00:00.11 . Memory (MB): peak = 3568.453 ; gain = 0.000 ; free physical = 5577 ; free virtual = 15788
INFO: [Timing 38-35] Done setting XDC timing constraints.
INFO: [Opt 31-138] Pushed 0 inverter(s) to 0 load pin(s).
Netlist sorting complete. Time (s): cpu = 00:00:00.07 ; elapsed = 00:00:00.07 . Memory (MB): peak = 3568.453 ; gain = 0.000 ; free physical = 5578 ; free virtual = 15790

Phase 1.2 IO Placement/ Clock Placement/ Build Placer Device
ERROR: [Place 30-640] Place Check : This design requires more RAMB36/FIFO cells than are available in the target device. This design requires 144 of such cell types but only 140 compatible sites are available in the target device. Please analyze your synthesis results and constraints to ensure the design is mapped to Xilinx primitives as expected. If so, please consider targeting a larger device.
ERROR: [Place 30-640] Place Check : This design requires more RAMB18 and RAMB36/FIFO cells than are available in the target device. This design requires 384 of such cell types but only 280 compatible sites are available in the target device. Please analyze your synthesis results and constraints to ensure the design is mapped to Xilinx primitives as expected. If so, please consider targeting a larger device.
ERROR: [Place 30-640] Place Check : This design requires more RAMB36E1 cells than are available in the target device. This design requires 144 of such cell types but only 140 compatible sites are available in the target device. Please analyze your synthesis results and constraints to ensure the design is mapped to Xilinx primitives as expected. If so, please consider targeting a larger device.
INFO: [Timing 38-35] Done setting XDC timing constraints.
Phase 1.2 IO Placement/ Clock Placement/ Build Placer Device | Checksum: 1497648bb

Time (s): cpu = 00:00:15 ; elapsed = 00:00:09 . Memory (MB): peak = 3568.453 ; gain = 0.000 ; free physical = 5480 ; free virtual = 15692
Phase 1 Placer Initialization | Checksum: 1497648bb

Time (s): cpu = 00:00:15 ; elapsed = 00:00:09 . Memory (MB): peak = 3568.453 ; gain = 0.000 ; free physical = 5480 ; free virtual = 15692
ERROR: [Place 30-99] Placer failed with error: 'Implementation Feasibility check failed, Please see the previously displayed individual error or warning messages for more details.'
Please review all ERROR, CRITICAL WARNING, and WARNING messages during placement to understand the cause for failure.
Ending Placer Task | Checksum: 1497648bb

Time (s): cpu = 00:00:15 ; elapsed = 00:00:09 . Memory (MB): peak = 3568.453 ; gain = 0.000 ; free physical = 5487 ; free virtual = 15699
INFO: [Common 17-83] Releasing license: Implementation
60 Infos, 21 Warnings, 0 Critical Warnings and 5 Errors encountered.
place_design failed
ERROR: [Common 17-69] Command failed: Placer could not place all instances
INFO: [Common 17-206] Exiting Vivado at Sun Apr 21 21:39:02 2019...
[Sun Apr 21 21:39:02 2019] impl_1 finished
wait_on_run: Time (s): cpu = 00:11:10 ; elapsed = 00:09:08 . Memory (MB): peak = 1700.316 ; gain = 0.000 ; free physical = 6770 ; free virtual = 16982
INFO: [Common 17-206] Exiting Vivado at Sun Apr 21 21:39:02 2019...
cp: cannot stat '/tmp/finn-build-_ub5ug/finnaccel/finnaccel.runs/impl_1/procsys_wrapper.bit': No such file or directory
Traceback (most recent call last):
  File "FINN/bin/finn", line 192, in <module>
    process_args(args)
  File "FINN/bin/finn", line 92, in process_args
    generate_hardware(net, dev, gen_bitfile=True)
  File "FINN/bin/finn", line 161, in generate_hardware
    ret.synthesis_bitfile()	
  File "/mnt/terabyte/pmousoul_data/Repos/FINN/FINN/backend/fpga/backend_util.py", line 71, in synthesis_bitfile
    subprocess.check_call(["sh", self.getBitfileSynthScriptPath()], cwd=self.path)
  File "/usr/lib/python2.7/subprocess.py", line 541, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['sh', '/tmp/finn-build-_ub5ug/make_pynq_bitfile.sh']' returned non-zero exit status 1
``

HPC-Ken, 2019-04-23:
Hello, 

So the placer has failed. You will need to scale back "BRAM_PROPORTION = 1" to a lower value, 0.9 or 0.8. 

https://github.com/Xilinx/FINN/blob/e6d44508b4f7c6ac8e2c6929f2ffa85ea5761708/FINN/core/config.py#L52

The hardware cost model does not match the output of the tools when using versions of Vivado later than those with which we built the tool.

maltanar, 2020-02-25:
I'm closing all issues relating to the v.01 version in preparation for the new release. Please note that v0.1 is now deprecated and unsupported.

closed 2020-02-25 by maltanar