Xilinx/finn #33: What level of precision does FINN support?

nmoroze, 2019-04-17:
Hi, I'm wondering what level of bit precision this version of FINN supports. I've noticed that a lot of the templates for various layer types are parameterized by the precision, but I recall reading in FINN-R that arbitrary bit precision was a new contribution and that code hasn't been released yet, implying that this FINN shouldn't support arbitrary precision.

HPC-Ken, 2019-04-18:
Hi,

So this version of FINN is from the original paper and only supports binary precision. We have not yet released the FINN-R version which for now we use in-house.

nmoroze, 2019-04-18:
Thanks! By binary precision, do you mean that weights and activations are represented by only 1 bit? If so, what are the precision parameters in some of the layer definitions controlling? For example, `WeightsPrecision` and `ActivationPrecision` in `matrix_multi_vector.h`?

HPC-Ken, 2019-04-19:
Hi, Yes, weights and activations are represented by 1 bit.

The additional parameters in the HLS code are indeed for higher precisions but in this release all the plumbing is not in place for end to end toolflow support for precisions other than binary. That came later. You could use the HLS library directly for non-binary networks if you're familiar with HLS programming, but I can't say I tested what we released here against non-binary parameters.

 When we release subsequent versions (no timeline as yet), those versions of the tool are built on top of a more advanced HLS library which supports the precisions found in our publications.

maltanar, 2020-02-25:
I'm closing all issues relating to the v.01 version in preparation for the new release. Please note that v0.1 is now deprecated and unsupported.

closed 2020-02-25 by maltanar