Xilinx/finn #34: reference design implementation failed

shuikeeper, 2019-04-22:
When I used "python FINN/bin/finn --device=pynqz1 --prototxt=FINN/inputs/cnv-w1a1.prototxt --caffemodel=FINN/inputs/cnv-w1a1.caffemodel --mode=synth" to try the framework with Vivado 2018.2, It reported place_design ERROR. 
log:
Phase 1.2 IO Placement/ Clock Placement/ Build Placer Device
ERROR: [Place 30-640] Place Check : This design requires more RAMB36/FIFO cells than are available in the target device. This design requires 144 of such cell types but only 140 compatible sites are available in the target device. Please analyze your synthesis results and constraints to ensure the design is mapped to Xilinx primitives as expected. If so, please consider targeting a larger device.
ERROR: [Place 30-640] Place Check : This design requires more RAMB18 and RAMB36/FIFO cells than are available in the target device. This design requires 384 of such cell types but only 280 compatible sites are available in the target device. Please analyze your synthesis results and constraints to ensure the design is mapped to Xilinx primitives as expected. If so, please consider targeting a larger device.
ERROR: [Place 30-640] Place Check : This design requires more RAMB36E1 cells than are available in the target device. This design requires 144 of such cell types but only 140 compatible sites are available in the target device. Please analyze your synthesis results and constraints to ensure the design is mapped to Xilinx primitives as expected. If so, please consider targeting a larger device.
INFO: [Timing 38-35] Done setting XDC timing constraints.
Phase 1.2 IO Placement/ Clock Placement/ Build Placer Device | Checksum: 1497648bb

Time (s): cpu = 00:00:22 ; elapsed = 00:00:15 . Memory (MB): peak = 3569.887 ; gain = 0.000 ; free physical = 7360 ; free virtual = 478279
Phase 1 Placer Initialization | Checksum: 1497648bb

Time (s): cpu = 00:00:22 ; elapsed = 00:00:15 . Memory (MB): peak = 3569.887 ; gain = 0.000 ; free physical = 7361 ; free virtual = 478279
ERROR: [Place 30-99] Placer failed with error: 'Implementation Feasibility check failed, Please see the previously displayed individual error or warning messages for more details.'
Please review all ERROR, CRITICAL WARNING, and WARNING messages during placement to understand the cause for failure.
Ending Placer Task | Checksum: 1497648bb

Time (s): cpu = 00:00:22 ; elapsed = 00:00:15 . Memory (MB): peak = 3569.887 ; gain = 0.000 ; free physical = 7367 ; free virtual = 478285
INFO: [Common 17-83] Releasing license: Implementation
60 Infos, 21 Warnings, 0 Critical Warnings and 5 Errors encountered.
place_design failed
ERROR: [Common 17-69] Command failed: Placer could not place all instances
INFO: [Common 17-206] Exiting Vivado at Mon Apr 22 05:28:05 2019...
[Mon Apr 22 05:28:06 2019] impl_1 finished
wait_on_run: Time (s): cpu = 00:13:35 ; elapsed = 00:15:15 . Memory (MB): peak = 1694.383 ; gain = 0.000 ; free physical = 7590 ; free virtual = 478508
INFO: [Common 17-206] Exiting Vivado at Mon Apr 22 05:28:06 2019...
cp: cannot stat '/tmp/finn-build-7wMqOe/finnaccel/finnaccel.runs/impl_1/procsys_wrapper.bit': No such file or directory
Traceback (most recent call last):
  File "FINN/bin/finn", line 192, in <module>
    process_args(args)
  File "FINN/bin/finn", line 92, in process_args
    generate_hardware(net, dev, gen_bitfile=True)
  File "FINN/bin/finn", line 161, in generate_hardware
    ret.synthesis_bitfile()	
  File "/app/FINN/FINN/backend/fpga/backend_util.py", line 71, in synthesis_bitfile
    subprocess.check_call(["sh", self.getBitfileSynthScriptPath()], cwd=self.path)
  File "/usr/lib/python2.7/subprocess.py", line 190, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['sh', '/tmp/finn-build-7wMqOe/make_pynq_bitfile.sh']' returned non-zero exit status 1

resource report post synthesis:
Resource | Utilization | Available | Utilization %
-- | -- | -- | --
LUT | 40210 | 53200 | 75.58
LUTRAM | 2148 | 17400 | 12.34
FF | 49600 | 106400 | 46.62
BRAM | 192 | 140 | 137.14
DSP | 38 | 220 | 17.27

By the way, the bin file FINN/bin/finn shows it supports VU9P device which has more BRAM resource as I know, but when I was tring to use this parameter, it didn't work.  The vivado-hls had created the VU9P BlackBoxjam IP successfully, but the "FINN/backend/fpga/scripts/make-pynq-vivado-proj.tcl" only support pynqz1.

HPC-Ken, 2019-04-23:
Different Vivado versions make different decisions about resource allocation. Your options are to use 2017.1 or to scale the usage of BRAM by scaling down the parameter "BRAM_PROPORTION = 1" https://github.com/Xilinx/FINN/blob/e6d44508b4f7c6ac8e2c6929f2ffa85ea5761708/FINN/core/config.py#L52

shuikeeper, 2019-04-24:
thanks for you reply. I tried to set BRAM_PROPORTION = 0.7/0.8, it didn't work. I am installing the 2017.1..

using the command "python FINN/bin/finn --device=vu9p --prototxt=FINN/inputs/cnv-w1a1.prototxt --caffemodel=FINN/inputs/cnv-w1a1.caffemodel --mode=synth" didn;t work. it seems the FINN framework only supports a PYNQ type FPGA board,  am I correct?

HPC-Ken, 2019-04-24:
For now we only support the PYNQ backend. The vu9p functionality hasn't been released. 

Are you still running out of BRAMS?

shuikeeper, 2019-04-24:
Yes, 2018.2 vivado reported the same utilization as before.

maltanar, 2020-02-25:
I'm closing all issues relating to the v.01 version in preparation for the new release. Please note that v0.1 is now deprecated and unsupported.

closed 2020-02-25 by maltanar