Xilinx/finn #35: Fresh docker install: synthesis example fails

Yaneke, 2019-04-30:
Hello, 

I have just installed the Docker image, strictly following the insctructions and then installed Vivado 18.3 in command line (https://www.xilinx.com/support/answers/70452.html).

Running the estimate command example works fine (`python FINN/bin/finn --device=pynqz1 --prototxt=FINN/inputs/lfc-w1a1.prototxt --mode=estimate`) but the synthesis one fails as shown here below:

```bash
root@1c985b9d0fd9:/app/FINN# python FINN/bin/finn --device=pynqz1 --prototxt=FINN/inputs/cnv-w1a1.prototxt --caffemodel=FINN/inputs/cnv-w1a1.caffemodel --mode=synth
WARNING: Logging before InitGoogleLogging() is written to STDERR
F0430 12:38:47.166438    36 upgrade_proto.cpp:95] Check failed: ReadProtoFromBinaryFile(param_file, param) Failed to parse NetParameter file: FINN/inputs/cnv-w1a1.caffemodel
*** Check failure stack trace: ***
Aborted (core dumped)
```

I performed this process twice to be sure I did not do something wrong the problem persists. What could this problem be?

Thanks

HPC-Ken, 2019-04-30:
Hi, Did you follow the step in the documentation to run git lfs pull?

Yaneke, 2019-04-30:
Oh right, I did not commit my docker before so it had not been registered!
Thank you for your help, I'm going to close the issue.

closed 2019-04-30 by Yaneke