Xilinx/finn #36: Python (PYNQ-Z1-Board) driver support for custom network

Chribecht, 2019-05-08:
Hi,
Beside this repository there is another FINN repository (BNN-PYNQ) with an overlay driver for the example networks. According to some issues here, it is possible to use the bitstream with the built-in weights in combination with the Python driver by commenting the loading of the parameters in the bnn.py module. So far so good.

Currently I have trained my own network (own topology) with HWGQ-Caffe and processed it with FINN. Currently I have the bitstream and tcl file belonging to the network. With the standard PYNQ classes I can also load these files onto the Zynq of the Z1 board, but without any driver support. In bnn.py dynamic libraries are loaded. I guess I can't use my network with these libraries. Any attempt to use the existing so-files with my own net where unsuccessful.

**Can this repo or the BNN-PYNQ repo be used to generate the required dynamic library using the "make-sw.sh" scripts? Is there a way to automatically generate the necessary PYNQ drivers for any network topologies?**

If you have any further questions, please do not hesitate to contact me and I look forward to your response.

maltanar, 2019-05-22:
Hi Chribecht,

The short answer is "yes", but we don't have a lot of documentation at the moment and this will involve lots of manual effort and digging. We're working on a cleaner FINN release with proper documentation and tutorials in the coming months.

If you have modified the topology in terms of how many inputs (e.g. pixels)/outputs (e.g. labels) it has, the prebuilt drivers and host code won't work correctly as you observed. The low-level drivers we use for both this repo and BNN-PYNQ are quite similar, so you should be able to use either. The BNN-PYNQ repo is a bit better organized so that would be my suggestion. You can find the low-level driver here, this gives a lightweight wrapper that lets you do register reads/writes and buffer copies:

https://github.com/Xilinx/BNN-PYNQ/tree/master/bnn/src/library/driver

What we call the "host code" calls the low-level driver and exposes higher level C++ function calls:

https://github.com/Xilinx/BNN-PYNQ/tree/master/bnn/src/library/host

Finally, there's a top-level wrapper that exposes the function calls to Python, which internally calls the host code. All three levels are used by the make-sw.sh to generate the shared libraries.

https://github.com/Xilinx/BNN-PYNQ/blob/master/bnn/src/network/cnvW1A1/sw/main_python.cpp

maltanar, 2020-02-25:
I'm closing all issues relating to the v.01 version in preparation for the new release. Please note that v0.1 is now deprecated and unsupported.

closed 2020-02-25 by maltanar