Xilinx/finn #38: LSTM with the new version of FINN

phillipgeier, 2019-10-04:
Hi,
I'm currently trying to implement an LSTM on FPGA using the "old" version (0.1) of FINN. I'm happy to see that the new version is on its way but I was wondering if the newer version will support LSTM networks. Having looked at brevitas and the finn-hlslib, there do not seem to be any references to LSTM or RNN networks in general.

Do you plan to support RNN / LSTM networks in the future?

Thank you for your time,

Phillip Geier

volcacius, 2019-10-04:
Hi Philip, LSTM are going to be supported on the Brevitas side at some point. They were in the original version (back when it was pytorch-quantization), but that was Pytorch 0.3.1 and everything has changed since then. The new implementation is going to be entirely JIT based (something like [this](https://pytorch.org/blog/optimizing-cuda-rnn-with-torchscript/)). On the FINN side, @giuliogamba is more suited to answer.

phillipgeier, 2019-10-04:
Thank you very much for your quick reply, glad to hear that LSTM are going to be supported on Brevitas !

maltanar, 2019-10-16:
Hi Phillip ,

Supporting LSTMs for FINN-style compilation is not on the development roadmap at the moment as we're just working on bringing up the infrastructure for regular feedforward nets. Conceptually it's possible to support them, but this isn't high priority for the FINN team at the moment. 

Is this something you are interested in? You may want to have a look at https://github.com/Xilinx/LSTM-PYNQ and check out [this paper](https://arxiv.org/pdf/1807.04093.pdf).

phillipgeier, 2019-10-24:
Yes I am interested in this. I had looked at the lstm-pynq repo and tried it out, and had a look at the paper but was curious about the new version of FINN :)
I understand that it is not a priority right now but I am still looking forward to this new version of FINN.

Than you for your work !

closed 2019-10-24 by phillipgeier