Xilinx/finn #39: Input data format

hornthomas, 2019-10-11:
Hi,

a while ago I downloaded what is now version 0.1 and finally found some time to have a look at it, only to find it is more or less deprecated by now. But I do hope you can still help me with me problem... 

Some Background: I tried to implement the CNV example on a ZED Board and to understand how to input the data to the network, I started with the generated HLS code and wrote a little function that reads in test data, creates the input data array and calls the generated function, everything still in software/C++. The program runs without errors, however the results I read back are always the same numbers, no matter if I provide real test image data or randomly generated values.

I had a closer look at the code and stumbled across this part in wrapper.h:

`
// convert to bipolar encoding, mapping {-1, 1} -> {0, 1}
stream<T> FPGABipolarConvThresholdLayer_0_bipolarenc;
ScaleShiftByConstant(singleInStrm, FPGABipolarConvThresholdLayer_0_bipolarenc, 0.500000f, 0.500000f, 3072);
// cast to single input element data type
stream<ap_uint<1> > FPGABipolarConvThresholdLayer_0_iconvert;
Cast(FPGABipolarConvThresholdLayer_0_bipolarenc, FPGABipolarConvThresholdLayer_0_iconvert, 3072);`

Can you please explain that? It seems as if the expected input (for CNV 32x32x3=3072 elements) is already constrained to values -1 and 1, instead of e.g. 8bit integers. From the papers I would have expected the input layer to handle integers instead of binarised values.

So my question is: What is the input data supposed to look like?

P.S.: In case you are wondering, that's the output values I read back each and every time:

> Reading output
> 0: -64
> 1: 106
> 2: 16
> 3: 34
> 4: -98
> 5: 86
> 6: 42
> 7: -76
> 8: 218
> 9: 76

maltanar, 2019-10-16:
Hi hornthomas,

The CNV network operates on RGB images and not binarized ones, we assume the initial input is quantized to 8 bits per color channel scaled to be in the range -1 to +1. All other intermediate activations and weights in the network are binarized, and the last layer produces 16 bits per class prediction. See following note from the FPGA'17 FINN paper:

> Note that the inputs to the first layer and the outputs from the last layer are not binarized; CNV accepts 32x32 images with 24 bits/pixel, and returns a 10-element vector of 16-bit values as the result.

Unfortunately we can't provide further support for v0.1 as we are focusing on building the new version. If you want to dig further into this yourself, the BNN-PYNQ host code for the CNV-w1a1 network may come in handy: https://github.com/Xilinx/BNN-PYNQ/blob/master/bnn/src/network/cnvW1A1/sw/main_python.cpp

hornthomas, 2019-10-22:
Hi @maltanar,

thank you for your reply.

Yes, I expected the input to be 3 x 8 bits/pixel, that's why I am wondering about this strange cast in the generated C++ file from ap_uint<64> to ap_uint<1>. In the generated code only 3 bits/pixel are fed into the CNV, and I can't see why.

maltanar, 2019-10-22:
Maybe I misunderstood your original question. This sounds like a bug; perhaps the input to the first layer is incorrectly marked as binary in the IR that is being fed to the code generator.

hornthomas, 2019-10-23:
Ah, okay, that could be possible. But given that this is one of your example nets, how did you get the results presented in the paper(s)? Or did you not use the version of FINN that was published on Github?

maltanar, 2019-10-23:
No, the results from the paper are from an internal version that is closer to what we have in BNN-PYNQ rather than the v0.1 branch here. If you're trying to reproduce some of the networks from the first FINN paper, the scripts in the [BNN-PYNQ](https://github.com/Xilinx/BNN-PYNQ/tree/master/bnn/src) repository are much more reliable and well-tested. 

I don't recommend using the v0.1 preview any further as it's likely that there are numerous bugs -- we're hoping to have the first release of the new version around December, and the development process is public this time (see the dev branch here on GitHub).

hornthomas, 2019-10-24:
Okay, thank you.

maltanar, 2020-02-25:
I'm closing all issues relating to the v.01 version in preparation for the new release. Please note that v0.1 is now deprecated and unsupported.

closed 2020-02-25 by maltanar