Xilinx/finn #43: Streamline input name mismatch in transformed matmul nodes

AndreaRigoni, 2020-02-11:
The Streamline transform generates strange names in inputs of the transformed MatMul nodes that are actually missing in the graph causing a disconnected network. 
For instance the following code generate a disconnected graph:
```python
    lfc = LFC(weight_bit_width=1, act_bit_width=1, in_bit_width=1)
    checkpoint = torch.load(trained_lfc_checkpoint, map_location="cpu")
    lfc.load_state_dict(checkpoint["state_dict"])
    bo.export_finn_onnx(lfc, (1, 1, 28, 28), export_onnx_path)
    model = ModelWrapper(export_onnx_path)
    model = model.transform(InferShapes())
    model = model.transform(FoldConstants())
    model = model.transform(GiveUniqueNodeNames())
    model = model.transform(GiveReadableTensorNames())
    model = model.transform(Streamline())
```
the corresponding net shown in netron is:
[image]

AndreaRigoni, 2020-02-12:
My fault .. I didn't get that some of the operands are not correctly shown in netron .. maybe because they are not correctly exported to onnx.

maltanar, 2020-02-18:
Hi Andrea, can you verify that this issue is resolved after PR #44 is merged?

AndreaRigoni, 2020-02-18:
Hi Yaman, verified ! Sorry, i missed to close the issue.

PR #44 (2020-02-17): [fix]  batchnorm transform always removes producer and consumer nodes
I think the intent of removing producer and consumer was limited to the squeeze and unsqueeze nodes but in reality the function was removing them in any case.

this should fix #43
closed 2020-02-18 by AndreaRigoni