Xilinx/finn #45: run-docker.sh cannot run if UNAME contains uppercase characters

0WN463, 2020-02-18:
A very small issue, but when running `sh run-docker.sh`, it failed because my username has uppercase characters. 

`invalid argument "finn_AdminCOOP" for --tag=finn_AdminCOOP: Error parsing refere
nce: "finn_AdminCOOP" is not a valid repository/tag: repository name must be low
ercase
See 'docker build --help'.
C:\Program Files\Docker\Docker\Resources\bin\docker.exe: Error parsing reference
: "finn_AdminCOOP" is not a valid repository/tag: repository name must be lowerc
ase.`

I fixed it by simply changing `DOCKER_UNAME=admin_coop` in run-docker.sh. Perhaps this case should be caught by run-docker.sh?

maltanar, 2020-02-18:
Hi @0WN463 -- thanks for reporting this, I just pushed a [https://github.com/Xilinx/finn/commit/86b209cf776acc9bc1afb18948190f9465ab544a](fix)to the dev branch. Can you confirm that this solves the problem for you?

0WN463, 2020-02-18:
Yea, that fixes the issue

closed 2020-02-18 by 0WN463