Xilinx/finn #46: output.npy not found in End-to-end-flow notebook

0WN463, 2020-02-28:
Under the section **Deployment and Remote Execution**, at 
```
import numpy as np
from finn.core.onnx_exec import execute_onnx
iname = parent_model.graph.input[0].name
oname = parent_model.graph.output[0].name
ishape = parent_model.get_tensor_shape(iname)
input_dict = {iname: x.reshape(ishape)}
ret = execute_onnx(parent_model, input_dict, True)
```

I got an error , 
```
FileNotFoundError                         Traceback (most recent call last)
<ipython-input-63-30954cff648b> in <module>
      5 ishape = parent_model.get_tensor_shape(iname)
      6 input_dict = {iname: x.reshape(ishape)}
----> 7 ret = execute_onnx(parent_model, input_dict, True)

/workspace/finn/src/finn/core/onnx_exec.py in execute_onnx(model, input_dict, return_full_exec_context)
    142         # topologically sorted
    143         for node in graph.node:
--> 144             execute_node(node, execution_context, graph)
    145     elif model_exec_mode == "remote_pynq":
    146         # use remote exec metadata built into model to execute on a remote PYNQ

/workspace/finn/src/finn/core/onnx_exec.py in execute_node(node, context, graph)
     47         sdp_node = getCustomOp(node)
     48         model = ModelWrapper(sdp_node.get_nodeattr("model"))
---> 49         ret = execute_onnx(model, context, True)
     50         context.update(ret)
     51     else:

/workspace/finn/src/finn/core/onnx_exec.py in execute_onnx(model, input_dict, return_full_exec_context)
    145     elif model_exec_mode == "remote_pynq":
    146         # use remote exec metadata built into model to execute on a remote PYNQ
--> 147         remote_exec(model, execution_context)
    148     elif model_exec_mode == "rtlsim":
    149         # use stitched IP for rtlsim

/workspace/finn/src/finn/core/remote_exec.py in remote_exec(model, execution_context)
     51     process_compile = subprocess.Popen(bash_command, stdout=subprocess.PIPE)
     52     process_compile.communicate()
---> 53     outp = np.load("{}/output.npy".format(deployment_dir))
     54     execution_context[model.graph.output[0].name] = outp

/opt/conda/lib/python3.6/site-packages/numpy/lib/npyio.py in load(file, mmap_mode, allow_pickle, fix_imports, encoding)
    420         own_fid = False
    421     else:
--> 422         fid = open(os_fspath(file), "rb")
    423         own_fid = True
    424 

FileNotFoundError: [Errno 2] No such file or directory: '/tmp/finn_own3d/pynq_deployment_473cmbj1/output.npy'
```

I could see that it is connected to the PYNQ board as `! sshpass -p {password} ssh {username}@{ip} 'ls -l {target_dir}/*'` list out the directory. When I look into the PYNQ directory, the output.py is indeed absent. I also read that the output.py should be generated by the driver.py.

maltanar, 2020-02-28:
If you ssh into the PYNQ board and do `sudo python3 driver.py` in the created deployment folder, do you get any error messages?

0WN463, 2020-02-29:
Just pulled from the latest master, problem still persists.

I'm a bit confused. I can't run the driver.py.
```
from pynq import Overlay
import numpy as np
from pynq import allocate
from finn.util.data_packing import (
    finnpy_to_packed_bytearray,
    packed_bytearray_to_finnpy
)
from finn.core.datatype import DataType
```

It requires imports from both pynq and finn. Are we supposed to install the finn packages onto the PYNQ board also?

maltanar, 2020-02-29:
No, this is meant to run on the PYNQ. Are you running this with either in `su` mode with `python3` or with `sudo python3` on the PYNQ board? You may want to familiarize with PYNQ a bit if you haven't already done so.

0WN463, 2020-02-29:
Oops. That's my bad. Have been using `python` instead of `python3`. Yea, driver.py managed to run successfully.

On another note, are we expected to run driver.py manually each time to generate output.npy? Or was it supposed to be part of the notebook flow (embedded in one of the transformations) but somehow it didn't run?

maltanar, 2020-02-29:
That's good to hear. As you guessed, there is remote execution built in and is part of the notebook. The credentials are put into the model with `DeployToPYNQ`, which is detected when you run `execute_onnx` on that model later on, in the Deploment and Remote Execution part of the notebook. Did you set the IP address, username and password of the PYNQ board correctly in the notebook? (cell 41) -- guessing this is not the problem since the files get copied to the board. You can try putting a pdb trace in https://github.com/Xilinx/finn/blob/master/src/finn/core/remote_exec.py#L35 somewhere and see if you can spot any problems.

maltanar, 2020-03-22:
I'm closing this issue since this seems to be resolved.

closed 2020-03-22 by maltanar

0WN463, 2020-03-30:
Yea, after numerous testing, I can't seem to reproduce it anymore. Still looking out for it though, if it were to happen