Xilinx/finn #68: Tests failing on random FC W1A1

volcacius, 2020-04-09:
Hello,

I'm testing FINN export and execution against Brevitas execution here: https://github.com/Xilinx/brevitas/blob/feature/finn_onnx_export_dev/test/finn_integration/test_finn_integration.py . Compared to the tests in FINN, input is randomized between 0 and 1.0, and I'm testing against the untrained version of the model too.
Tests on all the FC W1A1 topologies on untrained models fail:

```
test/finn_integration/test_finn_integration.py::test_brevitas_fc_onnx_export_and_exec[True-1-1-TFC] PASSED [  4%]
test/finn_integration/test_finn_integration.py::test_brevitas_fc_onnx_export_and_exec[True-1-1-SFC] PASSED [  8%]
test/finn_integration/test_finn_integration.py::test_brevitas_fc_onnx_export_and_exec[True-1-1-LFC] PASSED [ 12%]
test/finn_integration/test_finn_integration.py::test_brevitas_fc_onnx_export_and_exec[True-1-2-TFC] SKIPPED [ 16%]
test/finn_integration/test_finn_integration.py::test_brevitas_fc_onnx_export_and_exec[True-1-2-SFC] SKIPPED [ 20%]
test/finn_integration/test_finn_integration.py::test_brevitas_fc_onnx_export_and_exec[True-1-2-LFC] SKIPPED [ 25%]
test/finn_integration/test_finn_integration.py::test_brevitas_fc_onnx_export_and_exec[True-2-1-TFC] PASSED [ 29%]
test/finn_integration/test_finn_integration.py::test_brevitas_fc_onnx_export_and_exec[True-2-1-SFC] PASSED [ 33%]
test/finn_integration/test_finn_integration.py::test_brevitas_fc_onnx_export_and_exec[True-2-1-LFC] PASSED [ 37%]
test/finn_integration/test_finn_integration.py::test_brevitas_fc_onnx_export_and_exec[True-2-2-TFC] PASSED [ 41%]
test/finn_integration/test_finn_integration.py::test_brevitas_fc_onnx_export_and_exec[True-2-2-SFC] PASSED [ 45%]
test/finn_integration/test_finn_integration.py::test_brevitas_fc_onnx_export_and_exec[True-2-2-LFC] SKIPPED [ 50%]
test/finn_integration/test_finn_integration.py::test_brevitas_fc_onnx_export_and_exec[False-1-1-TFC] FAILED [ 54%]
test/finn_integration/test_finn_integration.py::test_brevitas_fc_onnx_export_and_exec[False-1-1-SFC] FAILED [ 58%]
test/finn_integration/test_finn_integration.py::test_brevitas_fc_onnx_export_and_exec[False-1-1-LFC] FAILED [ 62%]
test/finn_integration/test_finn_integration.py::test_brevitas_fc_onnx_export_and_exec[False-1-2-TFC] SKIPPED [ 66%]
test/finn_integration/test_finn_integration.py::test_brevitas_fc_onnx_export_and_exec[False-1-2-SFC] SKIPPED [ 70%]
test/finn_integration/test_finn_integration.py::test_brevitas_fc_onnx_export_and_exec[False-1-2-LFC] SKIPPED [ 75%]
test/finn_integration/test_finn_integration.py::test_brevitas_fc_onnx_export_and_exec[False-2-1-TFC] PASSED [ 79%]
test/finn_integration/test_finn_integration.py::test_brevitas_fc_onnx_export_and_exec[False-2-1-SFC] PASSED [ 83%]
test/finn_integration/test_finn_integration.py::test_brevitas_fc_onnx_export_and_exec[False-2-1-LFC] PASSED [ 87%]
test/finn_integration/test_finn_integration.py::test_brevitas_fc_onnx_export_and_exec[False-2-2-TFC] PASSED [ 91%]
test/finn_integration/test_finn_integration.py::test_brevitas_fc_onnx_export_and_exec[False-2-2-SFC] PASSED [ 95%]
test/finn_integration/test_finn_integration.py::test_brevitas_fc_onnx_export_and_exec[False-2-2-LFC] SKIPPED [100%]
```
This is on python 3.7, latest dev branch of finn, pytorch 1.1.0, on Windows 10.
If all tests were failing I would probably think that my environment is not supported, but given that most of them pass I'm suspicious the issue might be with FINN. 
Unfortunately my CI is broken at the moment, but you can reproduce it locally if you:
- have conda installed
- `pip install nox  pyyaml packaging` in any env you want, not necessarely conda (as long as conda is in your path)
- `git clone https://github.com/Xilinx/brevitas && git checkout feature/finn_onnx_export_dev`
- `cd brevitas && pip install -e .`
- `nox -v -s tests_brevitas_finn_integration-3.7(pytorch_1.1.0)`

Nox takes care of creating a matching conda enviroment and running the tests inside it.
Once the env is created, you can reuse it to rerun the tests with  `nox -v -r -s tests_brevitas_finn_integration-3.7(pytorch_1.1.0)`

volcacius, 2020-04-10:
Update: CI is back running, currently tests on Ubuntu fail for some other reason I need to debug, while Win and macOS show the behaviour I'm describing. Reference: https://github.com/Xilinx/brevitas/runs/575688333?check_suite_focus=true .

maltanar, 2020-04-14:
I'll look into this. Could you keep me posted if you do manage to reproduce this in Ubuntu in the meanwhile?

maltanar, 2020-04-15:
I'm having some trouble reproducing the testing environment for this:
```
root@a41954dbe415:/workspace/brevitas# nox -v -s tests_brevitas_finn_integration-3.7(pytorch_1.1.0)
bash: syntax error near unexpected token `('
```
Should the `(pytorch_1.1.0)` from the end of the `nox` command?

volcacius, 2020-04-15:
Sorry I forgot to specify you should put quotes around, so `nox -v -s "tests_brevitas_finn_integration-3.7(pytorch_1.1.0)"`.

maltanar, 2020-04-15:
I believe I've tracked down and fixed the issues with the integration test, opened a PR at https://github.com/Xilinx/brevitas/pull/116 -- in the end the issues were not in the codebase but rather on the Brevitas ONNX export logic for bipolar activations, plus a version issue with ONNX.

volcacius, 2020-04-16:
Can confirm it's now passing all the tests on my side on both 1.1.0 and 1.2.0 ([here](https://github.com/Xilinx/brevitas/runs/590703667?check_suite_focus=true)). Thanks

closed 2020-04-20 by maltanar