Xilinx/finn #72: Support parallelizing node-local transformations

maltanar, 2020-04-17:
Several FINN transformations like Vivado HLS IPGen and npysim compilation take a long time to run, which slows down development cycles. These transformations could be sped up considerably by running several nodes in parallel.

Although parallelizing general-purpose FINN graph transformations is a bit tricky due to potential sync issues, Vivado HLS IPGen etc. are all node-local transformations, i.e. the transform only accesses attributes from a single node, and modifies attributes on the same node as a result. These transformations are a lot easier to parallelize.

Here's a small proposal for getting there:

- Introduce a new kind of `Transformation` subclass called `NodeLocalTransformation`, with a new abstract class method called ´applyNodeLocal`.
- This ´applyNodeLocal` should take in a single ONNX node, and return a (possibly modified) single ONNX node. It should not have access to the entire `model` like the standard `Transformation`s do.
- The `apply` member of `NodeLocalTransformation` is responsible for parallel launching of tasks, something like:
```
pool = multiprocessing.Pool(NUM_DEFAULT_WORKERS)
old_nodes = model.graph.node
new_nodes = list(*pool.map(lambda x: self.applyNodeLocal(x), old_nodes))
model.graph.node = new_nodes
```

Once the basic infrastructure in place, we can start changing existing `Transformation` derivatives to be `NodeLocalTransformation` derivatives instead. Here is a (probaby incomplete) list of node-local transformations, most compute-heavy ones are marked with +:

```
CleanUp
CodeGen_ipgen +
CodeGen_npysim +
Compile ++
HLSSynth_IPGen +++
ReplaceVerilogRelPaths
SetExecMode
```

maltanar, 2020-04-17:
Tagging @HenniOVP who may be interested to help out on this.

HenniOVP, 2020-04-17:
That sounds like a very good proposal! I'll look into trying out this concept for the `HLSSynth_IPGen` transformation in the next few days.

One question which I have is if we really need to use the `multiprocessing` module or if just using the `multithreading` module would already be enough. Since in my experience the `multithreading` module is a bit more forgiving during development. For `HLSSynth_IPGen` most of the computing intensive time is spent outside of Python meaning that we don't get hindered by the global interpreter lock, but for other transformations this may be different. What is your opinion on this?

HenniOVP, 2020-04-20:
Hi,
I have made some progress with the parallelization. The HLSSynth_IPGen and the CodeGen_ipgen transformations are now working in the way, which you suggested.
The boost in performance is considerable, especially for the HLSSynth_IPGen. See this plot of execution times and speedups. (Note the log scale on the execution times.)
[image]

The reason why we only get a speedup between 2 and 3.5 is because the network has very few layers and some layers run significantly longer than others. Take a look at the annotated CPU time here:
[image]

So the parallelization would likely yield better results with the CNV network, which has more layers, instead of the FC networks. Do you know if there is currently a way to test this out? Since I could not get the CNV network to streamline or synthesize in the current release of FINN.

maltanar, 2020-04-20:
@HenniOVP that looks great! If you pull in the latest version of the `dev` branch you should be able to run the cnv-w1a1 network end to end, have a look at the `test/end2end/test_end2end_cnv_w1a1.py` -- this is quite fresh so please let me know if there are any problems. And it should indeed benefit more significantly from parallel synthesis due to more + bigger layers. We are also just putting the finishing touches on decoupled memory-compute support (weight memories instantiated as separate components and connected through streams), which may even out the differences between layers to some degree.

Do you want to send a pull request to `dev` so I can start reviewing and merging this?

HenniOVP, 2020-04-20:
Okay, I'll give the `test/end2end/test_end2end_cnv_w1a1.py` a try and report back.

Concerning the pull request: I will try to do this as soon as possible, but it will likely take me until Wednesday.

HenniOVP, 2020-04-20:
Quick update: The code from the `test/end2end/test_end2end_cnv_w1a1.py` script in the `dev` branch works nicely. Here is the updated result when running on eight cores on my Ryzen 2700X.
[image]

maltanar, 2020-04-21:
Great to hear the cnv-w1a1 worked out of the box as well, looks like more layers didn't necessarily help with the overall speedup though -- might be interesting to see if there is a single big layer that is taking longer than others and diminishing the parallel speedup returns.

If you'd like to experiment with `mem_mode="decoupled"` it should be just to change this line here, I just pushed a commit (9d7de20465b2833200cb4a10f1d5c661928a60a4) to actually take the parameter into account for cnv-w1a1 as well:

https://github.com/Xilinx/finn/blob/dev/tests/end2end/test_end2end_cnv_w1a1.py#L80

This may not work for cnv-w1a1 due to a limitation on the maximum decoupled memory size, which @auphelia is working on a fix for at the moment, but decoupled memory mode should already work for TFC and SFC.

HenniOVP, 2020-04-21:
Cool, I'll give that a try!
As for the low speedup in the CNV netowrk: It seems that there are actually two layers which are running for a very long time compared to all other layers. But I don't know exactly which two layers those were, probably the FC layers at the end of the network. I will try to make a detailed recording of CPU usage. Similar to what I did for the SFC network.

maltanar, 2020-04-22:
#78 is now merged -- thank you @HenniOVP for this contribution! I also took the liberty of converting the `Compile` transformation to a `NodeLocalTransformation`.

closed 2020-04-22 by maltanar

HenniOVP, 2020-04-23:
Great! I am glad that I could be of help.