Xilinx/finn #75: Transformation(s) to insert top-k nodes at the end of network

maltanar, 2020-04-20:
FINN should have a `Transformation` that inserts an [ONNX `TopK` node](https://github.com/onnx/onnx/blob/master/docs/Operators.md#topk) at the end of a classification network. This will pave the way for doing the top-K selection operation in hardware and reducing the output bandwidth requirements. Once #74 is resolved, the ONNX TopK node can be converted into the HLSCustomOp equivalent and be synthesized down to hardware.

maltanar, 2020-04-28:
@quetric expressed interest in working on this, but I couldn't assign him to this issue for some reason. Adding this comment here so that other interested parties can track the progress.

quetric, 2020-05-01:
Fixed in #84

maltanar, 2020-05-01:
#84 adds the transformation to insert the TopK node into the graph, we'll still need another `ConvertToHLSLayers`-type transformation that converts the `TopK` to `LabelSelect_Batch` once #74 is resolved.

closed 2020-07-29 by maltanar