Xilinx/finn #89: rtlsim fails when using Vivado 2019.2

quetric, 2020-05-05:
How to reproduce:

```
sh run-docker.sh
python setup.py test --addopts "-k test_fpgadataflow_fclayer_rtlsim --pdb"
```

Expected behaviour: test passes

Actual behaviour: test fails with an error in the pyverilator compile stage:
```
%Error: /tmp/finn_dev_lpetrica/code_gen_ipgen_StreamingFCLayer_Batch_0_vew52yqa/project_StreamingFCLayer_Batch_0/sol1/impl/verilog/StreamingFCLayer_Batch_0_StreamingFCLayer_Batch_0.v:118: Cannot find file containing module: regslice_both
```
Note that in contrast to 2019.1 and earlier, 2019.2 puts regslice modules in a separate verilog file.

quetric, 2020-05-08:
Debugged this further. Problem is caused by the naming of the verilog file containing regslice modules, which starting in 2019.2, is called `regslice_core.v`. Verilator assumes modules are found in (System)Verilog files of the same name, which is not the case here. 

The solution is to pass both the top Verilog file and `regslice_core.v` to verilator along with a `--top-module` argument indicating the name of the top module. However, PyVerilator does not support either passing of `--top-module` or providing multiple verilog files to Verilator.

quetric, 2020-05-08:
Submitted issue to PyVerilator: https://github.com/csail-csg/pyverilator/issues/10

maltanar, 2020-05-11:
Thanks for finding and fixing this @quetric. Changes have been merged into our fork of PyVerilator and I've updated `dev` to use the latest version in 5ab626ede5137df954a07c7c9989393170c6cb9e -- hopefully this fixes the issues you have observed with Vivado 2019.2

PR #104 (2020-05-20): Changed how pyverilator is called, to support Vivado 2019.2
This enables rtlsim with Vivado 2019.2 and fixes #89 

@maltanar I checked against several of the tests for dataflow ops (fclayer, maxpool, slidingwindow) and these pass. However, any block design TCLs which had been exported from Vivado 2019.1 may not work in 2019.2 and will need to be re-exported or modified. Any such cases should be picked up by deployment tests and I can fix the problematic TCLs if you point me to them.
closed 2020-05-21 by maltanar