Xilinx/finn #98: Transformation `FoldConstants()` raises `ONNXRuntimeError`

QDucasse, 2020-05-12:
Using the transformation `FoldConstants()` from  module `finn.transformation.fold_constants` raises an `ONNXRuntimeError`. The following script is run in a brand new FINNv0.3 docker container:
```python
import onnx
import brevitas.onnx as bo
from finn.util.test import get_test_model_trained
from finn.core.modelwrapper import ModelWrapper
from finn.transformation.infer_shapes import InferShapes
from finn.transformation.fold_constants import FoldConstants

tfc = get_test_model_trained("TFC", 1, 1)
bo.export_finn_onnx(tfc, (1, 1, 28, 28), "./tfc_w1_a1.onnx")
model = ModelWrapper("./tfc_w1_a1.onnx")
model = model.transform(InferShapes())
model = model.transform(FoldConstants())
```
It raises the following error:
```
...
[ONNXRuntimeError] : 2 : INVALID_ARGUMENT : Failed to load model with error: Unknown model file format version.
```
Full error message can be found here: https://pastebin.com/kG3rsYf1

As a consequence, all notebooks running this transformation cannot be run (`tfc_end2end_example`, `tfc_end2end_verification` and `cnv_end2end_example`, `1_brevitas_network_import`).

Testing the same piece of code in the docker container with the FINNv0.2.1b release raises no error.

maltanar, 2020-05-12:
Hi @QDucasse , thanks for reporting this. I'm able to run this snippet without any errors on `master@v0.3b`. Can you print out `onnx.__version__` and `onnxruntime.__version__` ? My container shows 

```
In [3]: onnx.__version__                                                                                                                 
Out[3]: '1.6.0'

In [4]: import onnxruntime                                                                                                               

In [5]: onnxruntime.__version__                                                                                                          
Out[5]: '1.2.0'
```

QDucasse, 2020-05-12:
I have the `1.7.0` version of `onnx` along with the `1.2.0` version of `onnxruntime`!

maltanar, 2020-05-13:
Good to hear that it was an easy fix. We've seen a few of these version incompatibility problems and it's probably a good idea to switch to fixed versions in the `requirements.txt`. Would you like to make a PR that specifies the package versions for `onnx` and `onnxruntime` in requirements.txt?

maltanar, 2020-05-13:
Actually I just saw that our Jenkins CI was failing due to this problem today, so decided to fix this quickly. 4436914f0a212584cbc36a6bae2381374031ff4b is now in both `dev` and `master` to address this. Can you confirm that the fix also works for you @QDucasse ?

QDucasse, 2020-05-14:
This solved the issue!

closed 2020-05-14 by QDucasse