Xilinx/finn #99: Conv Custom Image and Kernel Shape

SpontaneousDuck, 2020-05-12:
The Conv->Im2Col->slidingwindow conversion and the underlying hls code currently only support square kernels and images. 

I will be doing work on this issue and hopefully adding support in both the python and HLS code.

SpontaneousDuck, 2020-05-13:
SWU also does not work when kernel is the same height as the image. Looking in to that.

maltanar, 2020-05-13:
Great to hear, we've not had use cases with non-square images hence the current restrictions with square ones. I'm curious, is it image or non-image data use cases you have that require the non-square shapes?

SpontaneousDuck, 2020-05-13:
It is non-image data. Doing some research with a custom CNN on non-image binary data. Should not be too hard to change to my use case though! Data is still NCHW format before all the transformations.

maltanar, 2020-05-13:
OK, I'd start by forking `finn-hlslib` and adding support for non-rectangular SWU and a corresponding test. Best to make this a separate component since the SWU is a piece of critical infrastructure with lots of optimizations and my colleague who maintains hlslib would probably be a lot happier this way. Once the support for this is in place it's no issue at all to have an `if` in the codegen functions that calls a separate SWU for non-square 2D arrays.

SpontaneousDuck, 2020-05-13:
Sounds great! I would be way more comfortable with that so I don't potentially break the critical one. The ability to have different code will make it easier to implement I believe as well.

maltanar, 2020-12-17:
I'm closing this issue due to inactivity. Please feel free to reopen, or discuss further on the FINN gitter channel on https://gitter.im/xilinx-finn/community

closed 2020-12-17 by maltanar