Xilinx/finn #125: Modelwrapper helper to get a new unique tensor name

Tobi-Alonso, 2020-05-20:
Usually when doing transformation, new tensors must be created. 
A collision on tensor names will probably result in undesired behaviour 

I'm currently using:

```python
import finn.util.basic as util
def get_new_unique_tensor_name(model, max_trials = 10,random_stringLength = 6):
    model_tensor_names = model.get_all_tensor_names()
    for trials in range(max_trials):
        new_param_name = util.random_string(stringLength=random_stringLength)
        if new_param_name not in model_tensor_names:
            break
    else:
        raise Exception(
            "Not able to create new tensor name"
            + "after 10 trials. Net too big for the random tensor"
            + "name lenght chosen? Try larger stringLength?"
        )
    
    return new_param_name
```

having this as a helper may be useful

maltanar, 2020-05-20:
Hi Tobi, we have a method `ModelWrapper.make_new_valueinfo_name()` already that is used in numerous transforms. Does that suit your needs?

Tobi-Alonso, 2020-05-20:
Yaman, 
sorry, I didn't saw that one.
yes, is what I needed. Thanks.

closed 2020-05-20 by Tobi-Alonso