Xilinx/finn #154: Missing bitstring module in PYNQ deployment package

MicroRAsus, 2020-06-16:
I was following tfc_end2end_example.ipynb end-2-end example. In cell 43, I encountered an error when remote executing the inference on the PYNQ board.
[image]
I followed the trace to remote_exec.py. I was able to find the command line to execute the inference, so I manually executed it by logged into the PYNQ using ssh. It turns out the inference driver can't execute due to missing bitstring module. 
[image]
After downloading the bitstring module from https://github.com/scott-griffiths/bitstring and transfer it to PYNQ using scp command, I installed it using "python setup.py install". This fixed the issue.

QDucasse, 2020-06-22:
I ran into the same issue and this solution worked, thanks a lot!

maltanar, 2020-06-22:
Thanks for tracking this down and reporting this @MicroRAsus and good to hear @QDucasse that you had the same issue which is solved by this -- the PYNQ drivers we generate are indeed dependent on the `bitstring` module being installed. I installed it on mine way back and completely forgot that it's there :-)  btw this is also installable with e.g. `sudo pip install bitstring` on the PYNQ.

For now I've updated the Requirements section in the documentation to indicate that the bitstring module should be installed on the PYNQ: https://finn-dev.readthedocs.io/en/latest/getting_started.html#requirements

We could further automate this by installing the bitstring package via `ssh` onto the target board when doing the deployment, similar to what we do here for creating the target dir on the PYNQ:
 https://github.com/Xilinx/finn/blob/dev/src/finn/transformation/fpgadataflow/make_deployment.py#L82

If either of you would like to send a PR for this please feel free. I'm closing this issue since the fix is straightforward by installing this package on the board.

closed 2020-06-22 by maltanar

QDucasse, 2020-06-23:
I know this has been closed but I cannot manage to simply use `sudo pip install bitstring` (or even `pip3`) on the PYNQ as I get hit by a:
```bash
Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0xb56f3b30>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution',)': /simple/bitstring/
```
repeated four times followed by:
```bash
Could not find a version that satisfies the requirement bitstring (from versions: )
No matching distribution found for bitstring
```
Am I missing something or should I script a `curl` + `scp` version?

maltanar, 2020-06-23:
Is your PYNQ board normally able to connect to the internet? The error sounds like a connection problem. (btw yes it should be `pip3` and not just `pip` -- I'll fix that in the instructions).

```
xilinx@pynq-ci:~$ sudo pip3 install bitstring
Collecting bitstring
  Downloading https://files.pythonhosted.org/packages/c3/fc/ffac2c199d2efe1ec5111f55efeb78f5f2972456df6939fea849f103f9f5/bitstring-3.1.7.tar.gz (195kB)
    100% |████████████████████████████████| 204kB 3.4MB/s 
Installing collected packages: bitstring
  Running setup.py install for bitstring ... done
Successfully installed bitstring-3.1.7
```

QDucasse, 2020-06-23:
I am super dumb, I changed the configuration of my setup by connecting the PYNQ directly to my computer in ethernet since the going through the router was not super handy.... So it does not have any access to the internet.. Dumb question sorry... I will try to submit a PR when I reuse the correct setup!

ThomasCourtois, 2022-07-19:
I don't want to open a new issue because it seems to be the same but a file is missing and i don't understand why and how i can fix it
[image]