Xilinx/finn #166: LeNet5 and VGGNets cannot be exported to ONNX format

QDucasse, 2020-06-24:
`LeNet5` network (described here https://github.com/QDucasse/nn_benchmark/blob/master/nn_benchmark/networks/quant_lenet5.py) fails to export to ONNX from within the provided Docker container. The trace stack is the following:
```bash
Traceback (most recent call last):
  File "nn_benchmark/core/exporter.py", line 72, in <module>
    bo.export_finn_onnx(lenet5, (1,3,32,32), export_path_lenet5)
  File "/workspace/brevitas/brevitas/onnx/__init__.py", line 103, in export_finn_onnx
    torch.onnx.export(module, input_t, export_path, **torch_onnx_kwargs)
  File "/opt/conda/lib/python3.6/site-packages/torch/onnx/__init__.py", line 25, in export
    return utils.export(*args, **kwargs)
  File "/opt/conda/lib/python3.6/site-packages/torch/onnx/utils.py", line 131, in export
    strip_doc_string=strip_doc_string)
  File "/opt/conda/lib/python3.6/site-packages/torch/onnx/utils.py", line 369, in _export
    strip_doc_string)
RuntimeError: ONNX export failed: Couldn't export Python operator <python_value>

Defined at:
@torch.jit.script_method
def forward(self, value: torch.Tensor) -> torch.Tensor:
    value = self.forward_impl(value)
            ~~~~~~~~~~~~ <--- HERE
    return value

Graph we tried to export:
graph(%0 : Float(1, 3, 32, 32),
      %feature_extractor.0.weight : Float(6, 3, 5, 5),
      %feature_extractor.3.weight : Float(16, 6, 5, 5),
...
```
Full stack trace here: https://pastebin.com/Z5t4JU5D

---

The same type of error is noticeable when trying to export VGG (11/13/16/19 - https://github.com/QDucasse/nn_benchmark/blob/master/nn_benchmark/networks/quant_vggnet.py):
```bash
Traceback (most recent call last):
  File "nn_benchmark/core/exporter.py", line 74, in <module>
    bo.export_finn_onnx(vgg11, (1,3,32,32), export_path_vgg11)
  File "/workspace/brevitas/brevitas/onnx/__init__.py", line 103, in export_finn_onnx
    torch.onnx.export(module, input_t, export_path, **torch_onnx_kwargs)
  File "/opt/conda/lib/python3.6/site-packages/torch/onnx/__init__.py", line 25, in export
    return utils.export(*args, **kwargs)
  File "/opt/conda/lib/python3.6/site-packages/torch/onnx/utils.py", line 131, in export
    strip_doc_string=strip_doc_string)
  File "/opt/conda/lib/python3.6/site-packages/torch/onnx/utils.py", line 369, in _export
    strip_doc_string)
RuntimeError: ONNX export failed: Couldn't export Python operator <python_value>

Defined at:
@torch.jit.script_method
def forward(self, value: torch.Tensor) -> torch.Tensor:
    value = self.forward_impl(value)
            ~~~~~~~~~~~~ <--- HERE
    return value

Graph we tried to export:
graph(%0 : Float(1, 3, 32, 32),
      %features.0.weight : Float(64, 3, 3, 3),
      %features.1.weight : Float(64),
...
```
Full stack trace here: https://pastebin.com/A9kC03jm

---

In order to export the different models, I modified the Docker initialization by adding my repository (https://github.com/QDucasse/nn_benchmark) in the mix and run it from there. I ran the script in https://github.com/QDucasse/nn_benchmark/blob/master/nn_benchmark/core/exporter.py 

The quantized version of the MobilenetV1 can be exported without any issues with the same script and the corresponding network

maltanar, 2020-06-26:
Hey Quentin, thanks for reporting this. Since these are Brevitas examples, we'd like the export to be covered by the FINN test suite so let's start by adding export tests for these networks similar to this:

https://github.com/Xilinx/finn/blob/dev/tests/brevitas/test_brevitas_cnv.py

Could you make a new fork from the finn dev branch and add tests similar to the one above?

maltanar, 2020-12-17:
I'm closing this issue due to inactivity. Please feel free to reopen, or discuss further on the FINN gitter channel on https://gitter.im/xilinx-finn/community

closed 2020-12-17 by maltanar