Xilinx/finn #229: Invalid volume specification

libby28, 2020-09-16:
Unfortunately, I've been struggling to get my head around niche docker issues. 
When initially running run-docker.sh, everything is functional, but I then get a docker error as illustrated below.
Given it then mentions the path to the Vivado installation, I'm not sure what volume specification I'm meant to replace that with.
[image]

maltanar, 2020-09-16:
Hello,

Prior to running run-docker.sh, what is the value of your `VIVADO_PATH` environment variable? Based on the above it looks like it contains two copies of the same path whereas it should be `export VIVADO_PATH=/tools/Xilinx/Vivado/2019.1`

maltanar, 2020-12-17:
I'm closing this issue due to inactivity. Please feel free to reopen, or discuss further on the FINN gitter channel on https://gitter.im/xilinx-finn/community

closed 2020-12-17 by maltanar