Xilinx/finn #247: Connecting the FINN IP up to a microblaze

rachelglenn, 2020-11-05:
I am following the example tfc_endtoend in FINN. I am trying to export the IP and integrate it with the microblaze processor for an arbitrary board. Does anyone have any information of how to connect the IP generated from FINN with the microblaze?

maltanar, 2020-11-08:
Hi rachelglenn,

MicroBlaze supports streaming peripherals via what used to be called the FSL interface. You should be able to integrate a FINN-generated IPI design with a MicroBlaze using this interface, but we don't have any examples of this at the moment.

The rough workflow for testing this out for the tfc end2end example would be something like:

1. Follow the FINN flow in https://github.com/Xilinx/finn/blob/master/notebooks/end2end_example/tfc_end2end_example.ipynb until "3. Hardware Build"
2. Follow the steps under "Emulation of stitched IP" in https://github.com/Xilinx/finn/blob/master/notebooks/end2end_example/tfc_end2end_verification.ipynb to get a stitched IPI design. The `CreateStitchedIP` transformation there will create this under `/tmp/finn_dev_<username>/` (the exact path is available from the `vivado_stitch_proj` metadata property on the returned model.
3. Start a new hardware design with the MicroBlaze, import the stitched IPI design into it and connect them together.

rachelglenn, 2020-11-18:
Yaman thank you very much for your help. We were able to generate the IP by following your instructions. We are now trying to connect the IP up to the microblaze. 

We have tried connecting the microblaze M0_axis (AXI stream interface) to the input of finn IP neural network. We get the error:

[BD-41-237] Bus interface property TDATA_NUM_BYES does not match between finn_tfc_example_0/axis_0(49) and microblaze_0M0_AXIS(4)
[BD-41-237] Bus interface property TDATA_NUM_BYES does not match between finn_tfc_example_0/axis_0(4) and microblaze_0M0_AXIS(1)

The input to the neural network from endtoend_tfc is "Tdata_num_bytes =49". It seems as though the bus width or size doesn't match. We cannot figure out how to get them to match. 

Thanks,

-Rachel

maltanar, 2020-11-22:
Hi Rachel, glad to hear that there's been progress on this. It sounds like you're going to need AXI stream data width converters between the MicroBlaze and the FINN IP, these are available in Vivado IPI. 

For the input interface (the axis_0(49)) you may have to adjust the SIMD folding of the very first layer for the FINN-generated accelerator to make it divisible by the common bus widths; assuming this is for the MNIST examples (with 28x28=784 bytes per input for the examples we provide) you could set SIMD=16 to get a 16-byte input interface. Otherwise you could adjust the MicroBlaze AXI stream output to be 1-byte.

maltanar, 2020-12-17:
I'm closing this issue due to inactivity. Please feel free to reopen, or discuss further on the FINN gitter channel on https://gitter.im/xilinx-finn/community

closed 2020-12-17 by maltanar

rachelglenn, 2020-12-17:
Thank you for the help.