Xilinx/finn #255: TypeError: object of type 'NoneType' has no len() in Pre- and Porsprocessing

VanRillaerM, 2020-12-04:
Hi,
I am currently trying the FINN End to End Flow and I'm using the cnv_end2end_example.ipynb.
While running the first code block of the "Adding Pre- and Postprocessing" chapter, I came across this error:

```
/home/mikevr/.local/lib/python3.6/site-packages/finn/transformation/infer_data_layouts.py:113: UserWarning: Assuming 4D input is NCHW
  warnings.warn("Assuming 4D input is NCHW")

---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-7-490ae08423b8> in <module>
     14 # join preprocessing and core model
     15 pre_model = ModelWrapper(chkpt_preproc_name)
---> 16 model = model.transform(MergeONNXModels(pre_model))
     17 # add input quantization annotation: UINT8 for all BNN-PYNQ models
     18 global_inp_name = model.graph.input[0].name

~/.local/lib/python3.6/site-packages/finn/core/modelwrapper.py in transform(self, transformation, make_deepcopy, cleanup, fix_float64)
    116         while model_was_changed:
    117             (transformed_model, model_was_changed) = transformation.apply(
--> 118                 transformed_model
    119             )
    120         if cleanup:

~/.local/lib/python3.6/site-packages/finn/transformation/merge_onnx_models.py in apply(self, model)
    157         model = model.transform(InferShapes())
    158         model = model.transform(InferDataTypes())
--> 159         model = model.transform(InferDataLayouts())
    160         model = model.transform(GiveUniqueNodeNames())
    161         model = model.transform(GiveUniqueParameterTensors())

~/.local/lib/python3.6/site-packages/finn/core/modelwrapper.py in transform(self, transformation, make_deepcopy, cleanup, fix_float64)
    116         while model_was_changed:
    117             (transformed_model, model_was_changed) = transformation.apply(
--> 118                 transformed_model
    119             )
    120         if cleanup:

~/.local/lib/python3.6/site-packages/finn/transformation/infer_data_layouts.py in apply(self, model)
    124                 )
    125         for node in graph.node:
--> 126             graph_modified |= _infer_node_data_layout(model, node)
    127         return (model, graph_modified)

~/.local/lib/python3.6/site-packages/finn/transformation/infer_data_layouts.py in _infer_node_data_layout(model, node)
     90             # try to guess based on number of output dims
     91             for o in node.output:
---> 92                 ndims = len(model.get_tensor_shape(o))
     93                 model.set_tensor_layout(o, _dims_to_layout(model, node, ndims))
     94     # compare old and new output dtypes to see if anything changed

TypeError: object of type 'NoneType' has no len()
```
and this is the inside of my end2end_example/test directory:
[image]

Does anyone have an idea what could cause this issue?

maltanar, 2020-12-04:
Thanks for reporting this @VanRillaerM -- could you let me know which commit hash or branch of FINN you are getting this with?

VanRillaerM, 2020-12-04:
I am getting this with the master branch which I cloned a few days ago.

VanRillaerM, 2020-12-04:
[image]

maltanar, 2020-12-07:
I'm not able to reproduce this problem, unfortunately. Judging by the sizes of the generated files everything looks fine from the screenshot. 

Could you try:
1) adding the `--no-cache` flag to the line with `docker build` in `run-docker.sh` -- also watch out for errors in the log when the Docker container is first starting.

2) removing all the generated `*.onnx` files under that directory and re-running the notebook from the start

3) if that doesn't help, can you share a copy of the model at the point where the transformation fails? You can get a copy by doing the following:

```
# execute the failing jupyter notebook cell and get the error message
# add a new Python code cell 
import pdb
pdb.pm()
# once you are in the debug context (inside the debugger prompt)
print(node)
print(o)
model.save("debug.onnx")
```

This will print the name of the missing output and in which ONNX node, and it should give you a `debug.onnx` file we can use for further diagnosis.

VanRillaerM, 2020-12-07:
1. I'm not using Docker. I'm running Jupyter notebook in a Ubuntu 18.04 VM in Virtualbox. So I suppose changing the run-docker.sh file won't change anything. I tried it, just to be sure, it didn't work.

2. Unfortunately this also didn't change the error.

3. Okay this is the screenshot of the debug context
[image]

and here you have the debug.onnx file:
[debug.zip](https://github.com/Xilinx/finn/files/5655037/debug.zip)

maltanar, 2020-12-07:
You really should be using the provided `run-docker.sh`. Please note the following from the README:

`Due to the complex nature of the dependencies of the project, we only support Docker-based deployment at this time.`

We hope to have a `pip install`able FINN one day, but the nature of the dependencies (Vivado, Vivado HLS, verilator, dependencies on particular commit versions of Brevitas and finn-base) don't make this easy.

VanRillaerM, 2020-12-08:
Okay, thank you.
I've been trying it out. While running ./run-docker.sh I came across another problem.
[image]

Any idea what could cause this? I was following the instructions on https://finn.readthedocs.io/en/latest/getting_started.html#running-finn-in-docker

maltanar, 2020-12-08:
The FINN docker is designed to run without sudo/root, I suspect you may have run it with `sudo ./run-docker.sh` -- it's possible to set up Docker so that it doesn't require `sudo` and can just run with your regular user:

https://docs.docker.com/engine/install/linux-postinstall/

VanRillaerM, 2020-12-08:
The issue has been solved by using Docker. Thanks!

closed 2020-12-09 by VanRillaerM