Xilinx/finn #258: PermissionError: [Errno 13] Permission denied: '/tmp/finn_dev_mikevr/dataflow_partition0_c95xkldl'

VanRillaerM, 2020-12-09:
Hey, I'm stuck at 'creating a dataflow partition' in the tfc_end2end_example.ipynb.
It looks like there's no permission granted to go to use 'dataflow_partition0_c95xkldl'.

```
---------------------------------------------------------------------------
PermissionError                           Traceback (most recent call last)
<ipython-input-17-06e1a9637c66> in <module>
      2 
      3 model = ModelWrapper(build_dir+"/tfc_w1_a1_hls_layers.onnx")
----> 4 parent_model = model.transform(CreateDataflowPartition())
      5 parent_model.save(build_dir+"/tfc_w1_a1_dataflow_parent.onnx")
      6 showInNetron(build_dir+"/tfc_w1_a1_dataflow_parent.onnx")

/workspace/finn/src/finn/core/modelwrapper.py in transform(self, transformation, make_deepcopy, cleanup, fix_float64)
    116         while model_was_changed:
    117             (transformed_model, model_was_changed) = transformation.apply(
--> 118                 transformed_model
    119             )
    120         if cleanup:

/workspace/finn/src/finn/transformation/fpgadataflow/create_dataflow_partition.py in apply(self, model)
    110                 # save model
    111                 df_model_dir = make_build_dir(
--> 112                     "dataflow_partition" + str(target_partition_id) + "_"
    113                 )
    114                 df_model_filename = df_model_dir + "/df_model.onnx"

/workspace/finn/src/finn/util/basic.py in make_build_dir(prefix)
    147     try:
    148         inst_prefix = os.environ["FINN_INST_NAME"] + "/"
--> 149         return tempfile.mkdtemp(prefix=inst_prefix + prefix)
    150     except KeyError:
    151         raise Exception(

/opt/conda/lib/python3.6/tempfile.py in mkdtemp(suffix, prefix, dir)
    368         file = _os.path.join(dir, prefix + name + suffix)
    369         try:
--> 370             _os.mkdir(file, 0o700)
    371         except FileExistsError:
    372             continue    # try again

PermissionError: [Errno 13] Permission denied: '/tmp/finn_dev_mikevr/dataflow_partition0_c95xkldl'
```

When I go look at the tmp directory, I see that all permissions for tmp are given (777).
[image]

I've tried a few things with chmod or chown but it doesn't change a thing.
Any idea what could cause the error?

maltanar, 2020-12-09:
It's strange that it got that far without permission errors but fired up during dataflow partition creation... Could there be some leftover files in `/tmp/finn_dev_mikevr` that were created using `sudo ./run-docker.sh`? I would try stopping the Docker container, removing the entire `/tmp/finn_dev_mikevr` (reboot should also do it) then restarting it.

VanRillaerM, 2020-12-09:
I had to stop the container using ```systemctl restart docker```, not the way it's supposed to be done but ```docker stop finn_dev_mikevr``` wouldn't work because of an permission error
[image]

Aside from that...
- I deleted the /tmp/finn_dev_mikevr directory
- I deleted the finn directory (and cloned/installed it again)
after that I ran it again but the error is still the same

VanRillaerM, 2020-12-14:
The problem has solved itself.
I booted up my pc this morning, went through the steps of the notebook and there were no errors.

closed 2020-12-14 by VanRillaerM