Xilinx/finn #260: problem launching Jupyter notebook

srizwantariq, 2020-12-11:
Hello guys

Thanks alot for creating this framework. 
I started working on FINN. I followed the getting Started guide. 

I first ran this command  
`./run_docker.sh`

This is the how it executes 
[image]

Full Terminal Text is here.
[FINN_terminal.txt](https://github.com/Xilinx/finn/files/5679376/FINN_terminal.txt)

then i ran
`./run-docker.sh notebook` 
[image]

it didn't workout as explained in the getting started guide. 
`This will launch the Jupyter notebook server inside a Docker container, and print a link on the terminal that you can open in your browser to run the FINN notebooks or create new ones. .. note:: The link will look something like this (the token you get will be different)`

Could you please help me, how can i proceed from here onwards. 

Best regards
Rizwan

VanRillaerM, 2020-12-11:
- Hit "ctrl-D" to get out of your docker container
- run the command "./run-docker.sh notebook"

You can't use "./run-docker.sh notebook" when you're inside of your container 
(you can see that you are in Docker when your name is collored)

maltanar, 2020-12-11:
Hi Rizwan, the `run-docker.sh notebook` is to be used from outside the container as VanRillaerM pointed out above. Just an extra clarification:

The instructions in https://finn.readthedocs.io/en/latest/getting_started.html#running-finn-in-docker with titles "Getting an interactive shell for development or experimentation" and "Running the Jupyter notebooks" aren't meant to be followed after each other, they are two different modes.

srizwantariq, 2020-12-11:
Thanks alot @VanRillaerM  and @maltanar  :)

closed 2020-12-11 by srizwantariq