Xilinx/finn #264: fail to install finn

liyue2ppy, 2021-01-20:
**As I was installing FINN followed by [https://finn.readthedocs.io/en/latest/getting_started.html],  it wa failed with error "ERROR: Command "python setup.py egg_info" failed with error code 1 in /workspace/finn/",  I am using ubuntu18 and vivado2019.1, and the full error information is as follows:**
Obtaining file:///workspace/finn
    ERROR: Complete output from command python setup.py egg_info:
    ERROR: Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/workspace/finn/setup.py", line 50, in <module>
        setup(use_pyscaffold=True)
      File "/opt/conda/lib/python3.6/site-packages/setuptools/__init__.py", line 145, in setup
        return distutils.core.setup(**attrs)
      File "/opt/conda/lib/python3.6/distutils/core.py", line 108, in setup
        _setup_distribution = dist = klass(attrs)
      File "/opt/conda/lib/python3.6/site-packages/setuptools/dist.py", line 444, in __init__
        k: v for k, v in attrs.items()
      File "/opt/conda/lib/python3.6/distutils/dist.py", line 281, in __init__
        self.finalize_options()
      File "/opt/conda/lib/python3.6/site-packages/setuptools/dist.py", line 732, in finalize_options
        ep.load()(self, ep.name, value)
      File "/workspace/finn/.eggs/PyScaffold-3.2.3-py3.6.egg/pyscaffold/integration.py", line 105, in pyscaffold_keyword
        version_keyword(dist, keyword, setuptools_scm_config(value))
      File "/workspace/finn/.eggs/PyScaffold-3.2.3-py3.6.egg/pyscaffold/contrib/setuptools_scm/integration.py", line 17, in version_keyword
        dist.metadata.version = get_version(**value)
      File "/workspace/finn/.eggs/PyScaffold-3.2.3-py3.6.egg/pyscaffold/contrib/setuptools_scm/__init__.py", line 150, in get_version
        parsed_version = _do_parse(config)
      File "/workspace/finn/.eggs/PyScaffold-3.2.3-py3.6.egg/pyscaffold/contrib/setuptools_scm/__init__.py", line 113, in _do_parse
        "use git+https://github.com/user/proj.git#egg=proj" % config.absolute_root
    LookupError: setuptools-scm was unable to detect version for '/workspace/finn'.
    
    Make sure you're either building from a fully intact git repository or PyPI tarballs. Most other sources (such as GitHub's tarballs, a git checkout without the .git folder) don't contain the necessary metadata and will not work.
    
    For example, if you're using pip, instead of https://github.com/user/proj/archive/master.zip use git+https://github.com/user/proj.git#egg=proj
    ----------------------------------------
ERROR: Command "python setup.py egg_info" failed with error code 1 in /workspace/finn/
/tools/Xilinx/Vivado/2019.1/settings64.sh: line 5: /tools/Xilinx/DocNav/.settings64-DocNav.sh: No such file or directory
/tools/Xilinx/Vivado/2019.1/settings64.sh: line 7: /tools/Xilinx/SDK/2019.1/.settings64-SDK_Core_Tools.sh: No such file or directory

maltanar, 2021-01-21:
Could you try removing the old folder and get a fresh clone of the FINN repository from GitHub, with the latest (tag v0.5) master branch? This may be due to some leftover old egg files from the previous release.

liyue2ppy, 2021-01-29:
**Sorry but I have tried as what you said, that I removed the old folder and get the FINN repository with the tag v0.5. However, finally I get the same issue as follows:**
oh-my-xilinx @ 1bae737669901e762f581af73348332b5c4b2ada
Obtaining file:///workspace/finn
    ERROR: Command errored out with exit status 1:
     command: /opt/conda/bin/python -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/workspace/finn/setup.py'"'"'; __file__='"'"'/workspace/finn/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-05xx83wg
         cwd: /workspace/finn/
    Complete output (27 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/workspace/finn/setup.py", line 50, in <module>
        setup(use_pyscaffold=True)
      File "/opt/conda/lib/python3.6/site-packages/setuptools/__init__.py", line 145, in setup
        return distutils.core.setup(**attrs)
      File "/opt/conda/lib/python3.6/distutils/core.py", line 108, in setup
        _setup_distribution = dist = klass(attrs)
      File "/opt/conda/lib/python3.6/site-packages/setuptools/dist.py", line 444, in __init__
        k: v for k, v in attrs.items()
      File "/opt/conda/lib/python3.6/distutils/dist.py", line 281, in __init__
        self.finalize_options()
      File "/opt/conda/lib/python3.6/site-packages/setuptools/dist.py", line 732, in finalize_options
        ep.load()(self, ep.name, value)
      File "/workspace/finn/.eggs/PyScaffold-3.2.3-py3.6.egg/pyscaffold/integration.py", line 105, in pyscaffold_keyword
        version_keyword(dist, keyword, setuptools_scm_config(value))
      File "/workspace/finn/.eggs/PyScaffold-3.2.3-py3.6.egg/pyscaffold/contrib/setuptools_scm/integration.py", line 17, in version_keyword
        dist.metadata.version = get_version(**value)
      File "/workspace/finn/.eggs/PyScaffold-3.2.3-py3.6.egg/pyscaffold/contrib/setuptools_scm/__init__.py", line 150, in get_version
        parsed_version = _do_parse(config)
      File "/workspace/finn/.eggs/PyScaffold-3.2.3-py3.6.egg/pyscaffold/contrib/setuptools_scm/__init__.py", line 113, in _do_parse
        "use git+https://github.com/user/proj.git#egg=proj" % config.absolute_root
    LookupError: setuptools-scm was unable to detect version for '/workspace/finn'.
    
    Make sure you're either building from a fully intact git repository or PyPI tarballs. Most other sources (such as GitHub's tarballs, a git checkout without the .git folder) don't contain the necessary metadata and will not work.
    
    For example, if you're using pip, instead of https://github.com/user/proj/archive/master.zip use git+https://github.com/user/proj.git#egg=proj
    ----------------------------------------
WARNING: Discarding file:///workspace/finn. Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
/tools/Xilinx/Vivado/2019.1/settings64.sh: line 5: /tools/Xilinx/DocNav/.settings64-DocNav.sh: No such file or directory
/tools/Xilinx/Vivado/2019.1/settings64.sh: line 7: /tools/Xilinx/SDK/2019.1/.settings64-SDK_Core_Tools.sh: No such file or directory

**What can I do to solve this issue?**

liyue2ppy, 2021-01-30:
**I modify the script of setup.py， which replace the “setup(use_pyscaffold=True)”with “setup(use_scm_version=True)”，and I get the info as follows:**
Obtaining file:///workspace/finn
Collecting finn-base@ git+https://github.com/Xilinx/finn-base#egg=finn-base
  Cloning https://github.com/Xilinx/finn-base to /tmp/pip-install-9pohn55n/finn-base_5e9d7b6d893e4a29a17c4e7c33bf0983
  Running command git clone -q https://github.com/Xilinx/finn-base /tmp/pip-install-9pohn55n/finn-base_5e9d7b6d893e4a29a17c4e7c33bf0983
Requirement already satisfied: bitstring>=3.1.7 in /opt/conda/lib/python3.6/site-packages (from finn-base@ git+https://github.com/Xilinx/finn-base#egg=finn-base->finn==0.0.0) (3.1.7)
Requirement already satisfied: numpy in /opt/conda/lib/python3.6/site-packages (from finn-base@ git+https://github.com/Xilinx/finn-base#egg=finn-base->finn==0.0.0) (1.19.5)
Building wheels for collected packages: finn-base
  Building wheel for finn-base (setup.py) ... done
  Created wheel for finn-base: filename=finn_base-0.0.post0.dev96+g91fb606-py3-none-any.whl size=131667 sha256=6f362824921fa5662451b0fb972dbf87d719232ead3fc2952ccd52e091a649cf
  Stored in directory: /tmp/pip-ephem-wheel-cache-teiu2arl/wheels/33/4b/7f/eac7da20df3ef0145cd8dd97cd275584ed076b530fe2219ff2
Successfully built finn-base
Installing collected packages: finn-base, finn
  Attempting uninstall: finn-base
    Found existing installation: finn-base 0.0.post0.dev91+g1363981
    Uninstalling finn-base-0.0.post0.dev91+g1363981:
      Successfully uninstalled finn-base-0.0.post0.dev91+g1363981
  Running setup.py develop for finn
Successfully installed finn finn-base-0.0.post0.dev96+g91fb606

**When I check the version of finn， it shows：**
finn                          0.0.0                    /workspace/finn/src
**Dose this mean I installed it successfully？**

maltanar, 2021-02-02:
This is strange. Can you run the following commands in your FINN repository directory (on the host) and share the output?

* `git status` 
* `git tag` 
* `git describe`

liyue2ppy, 2021-02-06:
I think the issue is due to that I download the zip file of finn in github， instead of using “git clone”. And when the comand of "git clone" is used, the issue is solved. Thanks!

closed 2021-02-06 by liyue2ppy

Amur-Pal, 2022-07-22:
I used git clone yet still running into the same problem. Any help is welcome.

shahzaib3120, 2023-08-08:
Using git clone, getting the same error, I tried updating the setuptools_scm with pip now it gives a different error:

AttributeError: 'Configuration' object has no attribute 'parent'