Xilinx/finn #266: run-docker.sh does not mount $VITIS_PATH

michael-barrow, 2021-01-22:
Hello,

I have this issue with Ubuntu 20.04, Docker version 20.10.2, and Vitis 2020.2

I'm trying to run the docker image using the run-docker.sh script. like so:
./run-docker.sh

Unfortunately this does not work.

Docker reports that files within the folder $VITIS_PATH do not exist, but I can confirm they do in the native operating system. For example, the output log of run-docker.sh reports:

`...

/tools/Xilinx/Vivado/2020.2/settings64.sh: line 5: /tools/Xilinx/DocNav/.settings64-DocNav.sh: No such file or directory

...`

however a cat of the file reported non-existing yeilds:

`cat /tools/Xilinx/DocNav/.settings64-DocNav.sh
##############################################################
 Copyright (c) 1986-2020 Xilinx, Inc.  All rights reserved....

`

Any ideas?

michael-barrow, 2021-01-23:
I found that this is probably a Vitis issue. If I run 

`source /tools/Xilinx/Vivado/2020.2/settings64.sh`

I get the same error, regardless of trying to run the finn infrastructure or not.

michael-barrow, 2021-01-23:
I'm leaving the issue up in case someone unfamiliar with docker, like me, sees something similar.

My Vitis install seemed to have a different organization than expected by run-docker.sh
To fix this, I made sure to add a mount entry point in run-docker.sh with access to the paths that were not accessible.
The end of my run-docker.sh was modified like so:

```
…
  fi
  DOCKER_EXEC+="-v /tools/Xilinx:/tools/Xilinx "
  #DOCKER_EXEC+="-v $VIVADO_PATH:$VIVADO_PATH"
  DOCKER_EXEC+="-v $VITIS_PATH:$VITIS_PATH "
  DOCKER_EXEC+="-v $PLATFORM_REPO_PATHS:$PLATFORM_REPO_PATHS "
  DOCKER_EXEC+="-v $XILINX_XRT:$XILINX_XRT "
  DOCKER_EXEC+="-e VITIS_PATH=$VITIS_PATH "
  DOCKER_EXEC+="-e PLATFORM_REPO_PATHS=$PLATFORM_REPO_PATHS "
  DOCKER_EXEC+="-e XILINX_XRT=$XILINX_XRT "
  DOCKER_EXEC+="-e ALVEO_IP=$ALVEO_IP "
  DOCKER_EXEC+="-e ALVEO_USERNAME=$ALVEO_USERNAME "
  DOCKER_EXEC+="-e ALVEO_PASSWORD=$ALVEO_PASSWORD "
  DOCKER_EXEC+="-e ALVEO_BOARD=$ALVEO_BOARD "
  DOCKER_EXEC+="-e ALVEO_TARGET_DIR=$ALVEO_TARGET_DIR "
fi
DOCKER_EXEC+="$DOCKER_EXTRA "
DOCKER_EXEC+="$DOCKER_TAG $DOCKER_CMD"

$DOCKER_EXEC
```

closed 2021-01-23 by michael-barrow

maltanar, 2021-01-23:
Hi Michael. The `/.settings64-DocNav.sh: No such file or directory` message should be harmless by itself and should not prevent the Docker container from starting (since FINN doesn't use docnav). Did you encounter any other problems besides that message?