Xilinx/finn #267: Support for Vivado 20.2 and above

Jefferson111, 2021-01-24:
Vivado HLS has been phased out from 2020.2 and seems to be replaced by Vitis HLS. It breaks the code [here](https://github.com/Xilinx/finn-base/blob/c201c00a5e158550fea26cd004e7babc4a7f3002/src/finn/util/fpgadataflow.py#L67) when it attempts to search for vivado_hls instead of vitis_hls. Is there anyway to modify FINN to support this minor change or any known workaround?

maltanar, 2021-01-24:
Hi Jefferson11, Vitis HLS support is on the roadmap: https://github.com/Xilinx/finn/projects/1#card-47713963 and there's already a FINN hlslib branch in the works to make this happen: https://github.com/Xilinx/finn-hlslib/tree/vitis-hls but there's a few bits and pieces we need to put in and test. I'm hoping this will happen over the course of the next month or so.

Jefferson111, 2021-01-25:
Thank you and great to know! 😀 Look forward to the merging of the branch

closed 2021-01-25 by Jefferson111