Xilinx/finn #268: error with argid,when generate the xo file

simplelins, 2021-01-26:
Hi, 
   I had done a test for using the mobilenetv1 xclbin file with XRT, but when running time,the error reported for argid. and I modify the script 'gen_xo.tcl', when generate xo file.now I can run the mobilenetv1 xclbin file in ALVEO U200 with XRT. 
[image]
  when using XRT to run the xclbin file, we need the argid to be a sequense, please check! thanks!
[image]
  I also confused that 'Not applicable for AXI4-Stream ports' , but actually argid for 'AXI4-Stream ports' will cause the error!

quetric, 2021-01-26:
Hi @simplelins 

Which version of XRT are you using for deployment, and which version of Vivado/Vitis did you use for building the xclbin? Which branch of FINN did you use for the build?

simplelins, 2021-01-26:
Hi
   XRT：xrt_202010.2.6.655_18.04-amd64-xrt.deb https://www.xilinx.com/bin/public/openDownload?filename=xrt_202010.2.6.655_18.04-amd64-xrt.deb
  Vivado/Vitis：2020.1
  commit id for FINN:

> commit af783db8dc2a1d2e95bd569d39464b935520b6d2 (HEAD -> master, tag: v0.5b, origin/master, origin/HEAD)
> Author: Yaman Umuroglu <yamanu@xilinx.com>
> Date:   Thu Dec 17 23:08:20 2020 +0100_

    [Docs] bring back missing images

auphelia, 2023-02-14:
Hi @simplelins, could you please try if your issue persists with the latest FINN release?

auphelia, 2023-03-06:
I am closing this due to inactivity. Please feel free to reopen or create a new issue if the problem persists!