Xilinx/finn #271: Testfile cifar10-test-data-class3.npz can't be loaded via pk.resource_filename() instruction in cnv_end2end_example.ipynb notebook

chhait07, 2021-01-30:
Hi,
on executing line `fn = pk.resource_filename("finn", "data/cifar10/cifar10-test-data-class3.npz")` in the `cnv_end2end_example.ipynb` notebook i receive the error message `No such file or directory`. See screenshot.

[image]

I suppose the path should point to this file: https://github.com/Xilinx/finn/blob/master/src/finn/qnn-data/cifar10/cifar10-test-data-class3.npz . So, how do I have to change the instruction so that it works? I tried some stuff but nothing worked.

maltanar, 2021-02-01:
Hi, thanks for reporting this! Looks like a number of resource paths in notebooks weren't updated in v0.5, I just pushed a fix for this:

https://github.com/Xilinx/finn/commit/c69f4d30c03c24d844b33e6c4081f7db8f64a940

Essentially `pk.resource_filename("finn", "data/cifar10/cifar10-test-data-class3.npz")` needs to change to `pk.resource_filename("finn.data", "cifar10/cifar10-test-data-class3.npz")` for this case. The commit also includes a fix for #269 -- can you verify that this works for you as well?

chhait07, 2021-02-05:
Hi, 
I changed the line as you described above. Unfortunately it's still not working, see the following screenshot. The error says:
*FileNotFoundError: [Errno 2] No such file or directory: '/workspace/finn-base/src/finn/data/cifar10/cifar10-test-data-class3.npz'*

[image]

chhait07, 2021-02-14:
Hi,
I switched to branch c69f4d3 and tried again. This fixed it for me.

closed 2021-02-14 by chhait07