Xilinx/finn #275: execute_onnx() in tfc_end2end_example throws error

chhait07, 2021-02-05:
Hi everybody! I'm currently testing the [tfc_end2end_example](https://github.com/Xilinx/finn/blob/master/notebooks/end2end_example/bnn-pynq/tfc_end2end_example.ipynb) which crashes at the `execute_onnx()` command. There seems to be missing an output.npy file. 
Can you help me here? Is there a workaround for me, so that I can successfully execute the notebook?

[image]

chhait07, 2021-04-10:
Hi everybody, 
after updating setuptools by `sudo python3 -m pip install --upgrade pip setuptools` I can successfully execute the whole notebook including command `execute_onnx()`.
Thanks for your help!

closed 2021-04-10 by chhait07

ThomasCourtois, 2022-07-18:
it still doesn't work for me and I have no idea of the reason.. pls help me

julienmas, 2022-08-01:
> Hi everybody, after updating setuptools by `sudo python3 -m pip install --upgrade pip setuptools` I can successfully execute the whole notebook including command `execute_onnx()`. Thanks for your help!

Hi, did you updated setuptools on your PC or on the board?

julienmas, 2022-08-01:
Hi,
I have the exact same probem and updating setuptools didn't work for me.
```
xilinx@pynq:~$ pip3 install setuptools
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: setuptools in /usr/local/share/pynq-venv/lib/python3.8/site-packages (44.0.0)
WARNING: You are using pip version 21.2.1; however, version 22.2.1 is available.
You should consider upgrading via the '/usr/local/share/pynq-venv/bin/python3 -m pip install --upgrade pip' command.
```

I don't know if pip3 has to be updated, I tried to update it, it threw this error:
```
xilinx@pynq:~$ python3 -m pip install --upgrade pip
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: pip in /usr/local/share/pynq-venv/lib/python3.8/site-packages (21.2.1)
Collecting pip
  Downloading pip-22.2.1-py3-none-any.whl (2.0 MB)
     |████████████████████████████████| 2.0 MB 3.8 MB/s 
ERROR: Will not install to the user site because it will lack sys.path precedence to pip in /usr/local/share/pynq-venv/lib/python3.8/site-packages
WARNING: You are using pip version 21.2.1; however, version 22.2.1 is available.
You should consider upgrading via the '/usr/local/share/pynq-venv/bin/python3 -m pip install --upgrade pip' command.
```

Also I noticed that the notebook tries to reach `/tmp/finn_dev_julien/pynq_deployment_1g5ct2f0/output.npy`, but I can't see this folder on my board, only `/home/xilinx/finn_dev_julien/pynq_deployment_1g5ct2f0`, I don't know if this is normal. Anyway, output.npy doesn't exist there either:
```
xilinx@pynq:~/finn_dev_julien/pynq_deployment_1g5ct2f0$ ls
driver_base.py  finn       qonnx        resizer.hwh      validate.py
driver.py       input.npy  resizer.bit  runtime_weights
```

Does anyone has an idea what the problem is?

ThomasCourtois, 2022-08-01:
after a lot of research, for me the problem is about the version of the image in the PYNQ card. I had version 2.7 and it seems some path aren't the same as version 2.6. All good with 2.6 and waitting for some patch maybe

julienmas, 2022-08-01:
Thanks for your answer. I just tried to use PYNQ v2.6 instead of v2.7, but I stil have the same problem.