Xilinx/finn #279: FINN export with build_dataflow tool fails when NN model has 4b quantization

pgreksic, 2021-02-10:
This issue is created based on a question in the gitter channel of FINN project: [Gitter Question](https://gitter.im/xilinx-finn/community?at=601a01f0ae4b9b27c1965863)
I am creating this issue for easier tracking as requested by Mr. @maltanar, and also adding Mr. @HenniOVP per his own request.

# 1. Original question:
Hello, i was trying to export a modified version of Brevitas vgg-11-like cnv (from brevitas_examples/bnn_pynq) that i trained on a different dataset using the build_dataflow tool with custom folding config. I should also say that the neural network in question has 4 bit quantization on both weights and activations, and that the config is basically a slightly modified bnn_pynq/cnv_w2a2_folding_config from the finn_examples repository where i reduced PE and SIMD parameters until the network could fit on the PYNQ-Z1 board (based on build_dataflow estimates_output).

build_dataflow is being used from a modified jupyter inside a docker container (finn v0.5b, released on 17.12.2020.) which uses code from cybersecurity example notebooks and build.py for bnn_pynq from finn_examples repositroy, and so far i have been able to export 1 bit and 2 bit version of the same network using this code, Vivado version is 2020.1.

The build is successful when generating the stitched_ip, but fails with following error when generating bitfile, pynq_driver and deployment_package:
[image]

I have uploaded all outputs from the build (docker_outputs, build_dataflow outputs, folding_config and .onnx model) to:
https://github.com/pgreksic/FINN_Export_AXI_Lite_Missing

The only thing that i can find in outputs that seems to be related to the issue is unassigned addresses of AXI Lite interface in
finn_docker_output/vivado_stitch_proj__xg6ldie/vivado.log
[image]

Is there a way to solve this error and get the build working?

# 2. Additional input:
If it helps, while looking through intermediate models, i found that the model "splits" when creating dataflow partition:
[image]

The name of the model where "splitting" occurs first is 4_step_create_dataflow_partition.onnx and can be found in build_dataflow_output/output_final/intermediate_models/ directory of github repository, the issue persists in models further generated by the build_dataflow tool.

# 3. Additional notes:
  1. There is another matter that could be causing the issue, while training the NNs i have made a beginner mistake and used pytorch 1.7 with brevitas, thus the properties of my trained .onnx models are different from the releases in brevitas_examples directory of brevitas repository, but since FINN v0.5b uses "higher" properties and i was able to export networks with 1b and 2b quantization, i didn't think that was the cause and if it is I am sorry for wasting your time. The following model properties are generated by opening the models using the netron.app: https://netron.app/
     1. Model properties of cnv_w1a1 model are: {format: PyTorch v0.1.10}  
     2. Model properties of cnv_w1a1 when modified by FINN(version released before v0.5b)  are: {format: ONNX v4, producer: pytorch 1.1, imports: ai.onnx v9}
     3. My model properties are: {format: ONNX v6, producer: pytorch 1.7, imports: ai.onnx v9}
     4. My model properties after being modified by FINN v0.5b are: {format: ONNX v7, producer: fuse_model, imports: ai.onnx v12}
  2.  Because of the issue above if you try to perform FINN export with FINN version earlier than the v0.5b (released 17.12.20), the nodes will be of class:NoneType instead of class:list after trying to perform first "Tidy Up transforms"  in order to add pre and post processing to the starting model, which will cause an error, sry don't have picture of error. 
  3. While working on my own of verification notebook based on tfc_end2end_verification notebook i used the "manual export process" code from  cnv_end2end_example in order to prepare the model, FINN is still v0.5b, and the same "splitting" of model occurs after these transforms:
[image]

Again i have no picture proof for the claim under number 3 so use this info with a grain of salt, i will try to execute the code again and post the snapshot of the model splitting in the comments.

Note: I was able to execute the whole modified verification notebook for models with 1b and 2b quantization, there were a lot of warnings that i didn't analyze too much because the outputs matched and i had to set the value of LIVENESS_THRESHOLD to 10000000 (this might be too much but i haven't tested with lower values, but if left at default of 10000 an error occurs in the PyVerilator part of notebook).

fpjentzsch, 2021-02-11:
Hi, we've looked into this issue and identified a limitation in the streamlining transformations as the root cause:

If you observe the model graph for the 1b and 2b case, you will notice some scalar "Mul" nodes for quantization/scaling purposes that multiply by a fixed factor of 1. Streamlining removes these identity operations and further transformations proceed correctly.

In the 4b case, the same nodes exhibit a floating point value != 1, so they _should_ be handled by other streamlining steps to incorporate them into a subsequent MultiThreshold node. However, the limitation prevents the reordering of such nodes past the 2 MaxPool layers in the network. This is where the graph breaks into 3 separate dataflow partitions, which causes all following HLS transformations to fail.

I'm currently working on a fix for the streamlining transformation to support cases like this. In the meantime, you could stick with the unproblematic bit widths or work around this issue by modifying the network architecture. To this end, @maltanar suggested to place additional MultiThreshold nodes before the MaxPool.

pgreksic, 2021-02-11:
@fpjentzsch Thank you for answering in detail.
I am not sure should i close the issue now or not, so i will leave it open in case you need it to track this limitation in streamlining problem, but feel free to close it or leave it open as it suits you.

fpjentzsch, 2021-02-11:
@pgreksic Thank you for drawing our attention to this limitation! Let's leave this issue open for now.

fpjentzsch, 2021-02-23:
@pgreksic We've added an additional transformation to the streamlining process that should deal with the stuck Mul nodes.

You can try it out on the dev branch. We would appreciate any feedback as to whether this solves the issue for the model in question.

pgreksic, 2021-03-01:
@fpjentzsch Sorry for answering slowly.

I have tried exporting the model in question using the dev branch of finn and can confirm that the "splitting" no longer occurs, however now the export process fails with "Synthesis failed, no bitfile found" error because there isn't enough hardware resources on the PYNQ-Z1 board to implement the PYNQ driver, at least that is my suspicion. I have updated the github repository with new outputs: https://github.com/pgreksic/FINN_Export_AXI_Lite_Missing

To backup my claim of insufficient hardware resources, here are a few images:
1.  Image showing the mentioned export error:
[image]
2.  Image showing resource estimate output for brevitas_cnv_4w4a.onnx model:
[image]
I can't find the source of the info but i think PYNQ-Z1 has 280 or something similar of these BRAM18K modules, and here 344 are required.
3. And finally the contents of vivado.log file that point to my suspicion:
[image]

Direct link to vivado.log file where i found the error: [vivado.log](https://github.com/pgreksic/FINN_Export_AXI_Lite_Missing/blob/main/finn_dev_feedback_docker_output/vivado_zynq_proj_cr8hg2w8/vivado.log)

However, since this is more an issue of available hardware resources and since i haven't encountered the "splitting" model problem anymore in my limited testing, i think the this issue is resolved and i leave it to you to mark it as closed if you wish.

fpjentzsch, 2021-03-02:
@pgreksic Thanks for your detailed feedback. Since this is indeed a resource limitation, I'll close this issue.

To fit the model on your device, there is one more thing you could try (besides decreasing the model dimensions, bit widths, or folding factors). As LUTs don't seem to be the limiting factor here, you could experiment with the "ram_style" in your folding configuration and set more layers from "block" or "auto" to "distributed". This should force Vivado to implement the memory as LUTRAM instead of BRAM.

closed 2021-03-02 by fpjentzsch