Xilinx/finn #282: execute_onnx() in cnv_end2end_example throws error

chhait07, 2021-02-14:
Hi everybody,
on executing the cnv_end2end_example.ipynb I'm getting an error on function `execute_onnx()`. See the following screenshot. 
[image]
I got the same error on executing tfc_end2end_example as you can see [here](https://github.com/Xilinx/finn/issues/275)
In both cases the error messages say that an *output.npy* file is missing. 

Can you tell me why I'm getting this error?

maltanar, 2021-02-15:
I suspect the PYNQ environment variables (or the initial board/part setting) may be incorrect. Are you able to ssh xilinx@$PYNQ_IP into the board, and when you do, can you see the deployment folder created there? You can try to run the inference manually by invoking `sudo driver.py` in that folder.

A common cause for these errors is if you have forgotten to install the bitstring package on your PYNQ: `sudo pip3 install bitstring`

chhait07, 2021-02-18:
Hi,
yes I can log into my board and can see the created folder there. I tried to run the inference with no success, see the following screenshot:

[image]

These are lines 30 - 33 of driver.py:
`import argparse`
`import numpy as np`
`import os`
`from finn.core.datatype import DataType`

The packages argparse and numpy are installed on my board, the os package isn't. on `sudo pip3 install os` I'm getting the message:
*ERROR: Could not find a version that satisfies the requirement os
ERROR: No matching distribution found for os*

The bitstring package is installed on my board. I uninstalled it and re-installed it with no change in behavior.

Do I have to install a finn package (or something like that) on the board?

maltanar, 2021-02-18:
Correcting my instruction in the previous message (also edited now), can you try with:

`sudo python3.6 driver.py` 

(PYNQ has both python2 and python3, and the FINN drivers must be run with python3)

If this is the driver generated by the end-to-end flow you don't need to install any extra packages, the dependencies are included in the `finn/` subfolder there.

chhait07, 2021-02-20:
Hi,
I tried command `sudo python3.6 driver.py` getting this:
[image]

When I try the same command on the tfc_end2end_example I'm getting the same error followed by:
*ImportError: cannot import name 'allocate'*

chhait07, 2021-02-26:
Hi,
after updating my board from PYNQ version 2.4 to 2.6.0 I have now a different behavior. After `sudo python3.6 driver.py` I'm simply getting no output, see the following screenshot:
[image]

On executing the notebook again I now get a similar error than before: 
[image]

Do you have an idea about how to solve the problem?

chhait07, 2021-04-10:
Hi everybody, 
after updating setuptools by `sudo python3 -m pip install --upgrade pip setuptools` I can successfully execute the whole notebook including command `execute_onnx()`.
Thanks for your help!

closed 2021-04-10 by chhait07

julienmas, 2022-08-01:
Hi, did you run `sudo python3 -m pip install --upgrade pip setuptools` on your PC or on the board?