Xilinx/finn #288: Disjoint graph after streamline

dprzewlocka, 2021-02-23:
Hello!

Following the thread on Gitter, I am facing the problem with disjoint graph (three parts) after streamline. I am going to list all the steps that lead me to this result:

The network I am testing is built of the following layers: (Conv - BN - ReLU - MaxPool) - (Conv - BN - ReLU) - (Conv - BN - ReLU - MaxPool) - (Conv - BN - ReLU) - (Conv). Quantization for first and last convolutional layers is set to 8 bit int, for hidden layers 1 bit (binary) and for activation (ReLU) - 2 bits.

Next I perform Tidy-Up basing on cnv_end2end_example notebook:
```python
model = model.transform(DoubleToSingleFloat())
model = model.transform(InferShapes())
model = model.transform(FoldConstants())
model = model.transform(GiveUniqueNodeNames())
model = model.transform(GiveReadableTensorNames())
model = model.transform(InferDataTypes())
model = model.transform(RemoveStaticGraphInputs())
```

The onnx graph looks good after this operations. Next I add input quantization annotation:
```python
global_inp_name = model.graph.input[0].name
model.set_tensor_datatype(global_inp_name, DataType.UINT8)
model = model.transform(InferShapes())
model = model.transform(FoldConstants())
model = model.transform(GiveUniqueNodeNames())
model = model.transform(GiveReadableTensorNames())
model = model.transform(InferDataTypes())
model = model.transform(RemoveStaticGraphInputs())
```

With such model I move to streamlining:
```python
model = model.transform(MoveScalarLinearPastInvariants())
model = model.transform(Streamline())
model = model.transform(LowerConvsToMatMul())
model = model.transform(MakeMaxPoolNHWC())
model = model.transform(absorb.AbsorbTransposeIntoMultiThreshold())
model = model.transform(ConvertBipolarMatMulToXnorPopcount())
model = model.transform(Streamline())
model = model.transform(InferDataLayouts())
model = model.transform(RemoveUnusedTensors())
```
The graph looks like this:
<p align="center">
  [image]
</p>

Moving to Partitioning and Conversion to HLS Layers:
```python
# choose the memory mode for the MVTU units, decoupled or const
mem_mode = "decoupled"
model = model.transform(to_hls.InferBinaryStreamingFCLayer(mem_mode))
model = model.transform(to_hls.InferQuantizedStreamingFCLayer(mem_mode))
model = model.transform(to_hls.InferConvInpGen())
model = model.transform(to_hls.InferStreamingMaxPool())
model = model.transform(RemoveCNVtoFCFlatten())
# get rid of Tranpose -> Tranpose identity seq
model = model.transform(absorb.AbsorbConsecutiveTransposes())
# infer tensor data layouts
model = model.transform(InferDataLayouts())
parent_model = model.transform(CreateDataflowPartition())
sdp_node = parent_model.get_nodes_by_op_type("StreamingDataflowPartition")[0]
sdp_node = getCustomOp(sdp_node)
dataflow_model_filename = sdp_node.get_nodeattr("model")
# save the dataflow partition with a different name for easier access
dataflow_model = ModelWrapper(dataflow_model_filename)
```

The final model looks like this:
<p align="center">
  [image]
</p>

I tested this on the dev branch (the presented output) as well as master, the result is similar for both cases. What could be the cause?

Thank you.

maltanar, 2021-02-23:
Hi dprzewlocka, the culprit seems to be the MaxPool layers not getting converted to HLS. `model.transform(to_hls.InferStreamingMaxPool())` should normally take care of this, but it looks like it's not finding the pattern that it expects. 

At a glance I can't see what would be the problem so this needs a closer look. Are you able share the ONNX file for your model? (should be possible to attach it as a file to the issue here). 

If not, or if you want to try debugging this yourself, have a look at the `InferStreamingMaxPool` source https://github.com/Xilinx/finn/blob/master/src/finn/transformation/fpgadataflow/convert_to_hls_layers.py#L161 and maybe you can spot the pattern mismatch. For instance, there's a check for `ifm_dim % k == 0`, otherwise the HLS conversion there will fail. If that is the problem, we may be able to provide a workaround.

dprzewlocka, 2021-02-24:
Hi @maltanar, thank you very much, indeed the problem was in converting the MaxPool layer. The `ifm_dim % k` was not equal to `0` and the conversion failed. For my case I just had to slightly modify the input's size, so the shape of each filter (conv and maxpool) matches it's input. The onnx dataflow model looks great now. However I think it's worth noting that conversion of such architecture works on dev branch right now (fails on master despite the accurate shapes).

Therefore no workaround is needed for me and I close the issue.

closed 2021-02-24 by dprzewlocka