Xilinx/finn #296: usage error

simplelins, 2021-03-05:
transformed_model
  File "/finn/src/finn/transformation/fpgadataflow/insert_fifo.py", line 89, in apply
    fld_shape_2 = n1.get_folded_input_shape(ind=idx)
TypeError: get_folded_input_shape() got an unexpected keyword argument 'ind'

https://github.com/Xilinx/finn/blob/af783db8dc2a1d2e95bd569d39464b935520b6d2/src/finn/transformation/fpgadataflow/insert_fifo.py#L89

fpjentzsch, 2021-03-10:
This is likely related to some residual connection (join node) in your graph. I assume you are experimenting with the work-in-progress Resnet-50 example?

simplelins, 2021-03-10:
Yeah, you are right. I am working with resnet50. I had checked the source code, and I can not find any function that named 'get_folded_input_shape' with a parameter.  
and for resnet50, are there  any news I can learn?

maltanar, 2021-03-11:
The ResNet-50 streamlining etc. will require a number of updates and custom transformations so please wait for the official release. We're hoping to get everything wrapped up over the next month.

auphelia, 2023-02-14:
I am closing this issue because it does not apply to the current release. Please use version v0.9.

closed 2023-02-14 by auphelia