Xilinx/finn #298: Network architecture - spatial dim in conv-> fc transition

dprzewlocka, 2021-03-16:
Hi,

following the issue mentioned in the gitter channel, I am facing the problem with custom network architecture (I'll repeat the description and add some new information).
My network is build of two conv layers and two fc layers (with bn, pool and activation between). During the streamlining, precisly the `AbsorbTransposeIntoMultiThreshold()` I got the assertation error `Untested spatial dim in conv->fc transition, proceed with caution!` (file finn/src/transformation/streamline/absorb.py, around line 355). In Brevitas/FINN examples the network CNV1w1a has subsequent layers build in such way, that the size of feature maps from the last convolutional layer is equal to 1. So I assumed this is the "tested spatial dim in conv->fc transition" - my network does not have this feature. So I commented out the assertation (as I decided to check it anyway). 
During hardware generation, I got the following error (file src/finn/transformation/fpgadataflow/insert_fifo.py, around line 98):
`The folded output shape of the first node is not the same as the folded output shape of the second node. A streaming fifo can't be implemented in between these nodes.`

After quick debugging it appeares that this situation is perhaps connected with layers last convolutional and first fully connected.
After the advice from @maltanar I decided to comment out this assertation and check what will happen. The hardware generation is now finishing without errors. I succesfuly run the network in the hardware, however output isn't correct.

I decided to check the Python simulation of the streamlined onnx model and the output matches the one obtained from hardware, but not the golden (Brevitas) one. What is more, the difference is already visible after first maxpool layer. I am following strictly the notebook cnv_end2end - that is, I also add the pre- and postprocessing (the Brevitas model accepts input in [0,1] range, the streamlined one includes division by 255).

I am not sure whether this is connected to this architecture issue, but until now I haven't found any solution. If it would be of any help, I can share the network's architecture. I'm also attaching the onnx model - the one obtained from the Brevitas model and the streamlined one (had to manualy add .zip to the name of the files, should work fine after renaming it back).

Thank you!

[XNOR-V3-T-Brevitas-Raw.onnx.zip](https://github.com/Xilinx/finn/files/6148705/XNOR-V3-T-Brevitas-Raw.onnx.zip)
[XNOR-V3-T-streamlined.onnx.zip](https://github.com/Xilinx/finn/files/6148769/XNOR-V3-T-streamlined.onnx.zip)

maltanar, 2021-03-23:
Thanks for taking the time to describe the issue in detail and sharing the ONNX files Dominika, I'm able to reproduce this. This is a limitation that we'd like to fix and I'm hoping we can get this fixed in the next few weeks.

maltanar, 2021-05-18:
Hi @dprzewlocka , PR #326 is now merged, so if you try again this use-case with the latest `dev` branch this issue should be fixed.

closed 2021-05-28 by maltanar