Xilinx/finn #304: Fully convolutional networks

dprzewlocka, 2021-04-07:
Hi! Following the conversation from gitter channel, I came across the issue with fully convolutional networks - hope the description below is enough to reproduce the problem, but if you need more information I'll be happy to share.

The problem origins from a bigger architecture (with several convolutional, batch normalization, pooling and activation layers), but for simplicity I decided to run some tests using the following network description:

```
class QNet(nn.Module):
    FIRST_LAYER_QUANT_TYPE = QuantType.BINARY
    FIRST_LAYER_BIT_WIDTH = 1
    WEIGHT_QUANT_TYPE = QuantType.BINARY
    WEIGHT_BIT_WIDTH = 1
    ACTIVATION_BIT_WIDTH = 1
    ACTIVATION_QUANT_TYPE = QuantType.BINARY

    def __init__(self, input_channels=1):        
        super(QNet, self).__init__()
        self.conv1 = qnn.QuantConv2d(input_channels, 64, 5,
                                     weight_quant_type=self.FIRST_LAYER_QUANT_TYPE,
                                     weight_bit_width=self.FIRST_LAYER_BIT_WIDTH,
                                     bias=True)

        self.clip_weights(-1, 1)

    def clip_weights(self, min_val, max_val):
        self.conv1.weight.data.clamp_(min_val, max_val)

    def forward(self, x):        
        CONV1 = self.conv1(x)
        output = CONV1  

        return output

net = QNet()

cnv = net.eval()

input_shape = (1, 1, 32, 32)
bo.export_finn_onnx(cnv, input_shape, "Brevitas-Raw.onnx")
```

For hardware generation (and previous steps) I follow the exemplar notebook - with the only difference in lack of postprocessing (TopK) for obvious reasons. The network's input should be in range [0, 1], so I add the preprocessing (division by 255). The generation process itself runs without any errors/warnings.
For easier debugging, for input I use the array with all values set to 255 (for hardware, Python and C++ simulations; for Brevitas - all set to 1).

The hardware output (after multiplication and addition operations that are not performed in FPGA) doesn't match the one obtained from Brevitas. 
But: the **Python simulation** (using streamlined model)  **and the C++ simulation** (using folded and then integrated model - so I think no extra operations are needed) **match the hardware output**.

Perhaps I'm missing something? Or is the fully convolutional architecture not supported?

Thank you!

jalezeta, 2021-04-09:
Hi there Dominika! It's a pleasure to help you.

I have tried to reproduce your issue. First, I have executed the simple CNV model in Brevitas with an input tensor (all values set to 1). The output array is filled with positive and negative floating-point values between -2 and 1 (approximately). Then, I have exported the simple CNV model to FINN ONNX (from brevitas). I have performed the tidy-up transformations and added the preprocessing (Div by 255). Now I execute the ONNX model where the input array has all values set to 255. The output array contains only 0 values (compared to the positive and negative values seen in brevitas). Is this what you are experiencing? Could you share the code of your tests? That can help me reproducing the error and diving deeper into what is causing it.

dprzewlocka, 2021-04-09:
Hi, thank you for your time! I'm attaching the notebook with all the code (had to add .zip for github to accept the file, should work after renaming it back). Sorry to be obvious here :), but if you're gonna run it make sure to update the values for multiplication and addition for the hardware output.

[fully_convolutional_tests.ipynb.zip](https://github.com/Xilinx/finn/files/6285238/fully_convolutional_tests.ipynb.zip)

Actually the results from my experiments are different: the values from Brevitas, simulations and hardware are all "reasonable" - that is, positive and negative floating-point - they just don't match.

jalezeta, 2021-04-21:
Hi Dominika, first of all, sorry for the late response. Thank you for sending your testing notebook, that helped me a lot in understanding your problem and replicating the error. Let me explain some things that I have experienced:

1. I have seen the results in C++ simulation, python simulation & hardware results are matching, as you described earlier. This indicates that FINN is consistent in the three different methods you tested yourself.
2. FINN applies step-by-step transformations into the initial FINN-ONNX to finally generate the hardware.
3. The first thing here is to execute the model at various stages of the FINN workflow and go from bottom to top until you Identify where the error is being introduced. 

The very first transformation after Brevitas -> FINN export is the `InferShapes()` transformation, as you perform in your notebook. 
 Based on this, I could see that the result after `InferShapes()` is matching the Python sim, C++, and HW results. This shows that FINN is not introducing any error, and the result is being consistent from top to bottom.  The problem here is that the Brevitas golden model is not matching what FINN is interpreting.

I saw that you are enabling the bias. I noticed that the brevitas golden model output and FINN output **do match** if you set the `bias=False`. This might be a workaround for your bigger network. The bias should get canceled out if you are applying batchnorm after the conv layer. 

Let me investigate deeper until I find the root cause of this problem. I will get back once I know more about this issue.

dprzewlocka, 2021-04-26:
Hi, thank you for your answer. Following your advice, I managed to successfully run the convolutional layer with bias set to False: all outputs match the golden model output. 
Then I performed two more experiments: (1) I set the bias to True and added the Batch Normalization layer and (2) I set bias to False and added the Batch Normalization layer. For both experiments the problem is repeated: the Brevitas golden model is not matching what FINN is interpreting. So the question is: in case of fully convolutional networks, is it necessary to disable bias and batch normalization?

maltanar, 2021-04-29:
Hi Dominika, to give an update on this: the bias export is likely an export bug and something I'm working on addressing. 

The bias=False case with batchnorm should work correctly though. One thing to be aware of is to make sure you've put the network into evaluation mode `pytorch_model.eval()` before doing the export and also for generating the golden result, since batchnorm changes behavior.

If you still observe this problem after doing `.eval()` could you provide a minimal testcase similar to e.g. https://github.com/Xilinx/finn/blob/master/tests/brevitas/test_brevitas_QConv2d.py so I can try to reproduce the exact problem?

dprzewlocka, 2021-04-30:
Hi Yaman, the scenario with batch normalization and ``bias=False`` works fine - I'm not sure why I received different results before. Thank you for your help - I'll be waiting for more info on the bias problem :)

maltanar, 2021-05-28:
The `bias=True` cases (as well as quantized biases) should now all work correctly after PR #339 so closing this issue. Dominika, please give this a try and feel free to reopen the issue if the problem hasn't been fixed for your use case.

closed 2021-05-28 by maltanar

dprzewlocka, 2022-02-18:
Hi Yaman! Should the network, let's say, build of one conv2d layer with fp32 bias and no batch normalization after work in FINN 07?